zephyrproject-rtos/zephyr #16365: lwm2m: enable with CONFIG_NET_RAW_MODE

erkki, 2019-05-23:
Right now there's code in lwm2m_engine.c that depends on CONFIG_NET_IPV4 &/ CONFIG_NET_IPV6 defined to work.

This is problematic for platforms relying on CONFIG_NET_RAW_MODE as that is defined as 

Here is what I had to do to get a proof-of-concept compile under nordic 9160dk: https://github.com/bear-metal/zephyr/tree/poc_lwm2m_net_raw

I'm not sure what a good direction to go would be in terms of preferred code structure (introduce a new Kconfig var? Make things more granular & pick what's needed?) but happy to help put together a patch &/ dig further with some guidance from upstream.

I believe @NordicPlayground is also looking into lwm2m support so maybe good to @/cc someone from there.

jukkar, 2019-05-23:
The CONFIG_NET_RAW_MODE is basically meant for cases where there is no upper part of the network stack. Its help text says
```
This is a very specific option used to built only the very minimal
part of the net stack in order to get network drivers working without
any net stack above: core, L2 etc... Basically this will build only
net_pkt part. It is currently used only by IEEE 802.15.4 drivers,
though any type of net drivers could use it.
```

So if you enable this option, then basically you cannot use IPv4 which means that LWM2M cannot be used. What kind of use cases you have that you need to enable this RAW option?

erkki, 2019-05-24:
Hi @jukkar, nrf9160_pca10090ns uses nordics bsdlib (which depends on CONFIG_NET_RAW_MODE) to implement a sockets api on top of the modem which implements an IP stack internally.

But the IP util functions are still convenient to have in userspace for shuffling around addresses and such. And the lwm2m client lib depends on some of the config macros being defined which triggers too much inclusion at other places.

I got the non-dtls non-bootstrap mode lwm2m client example working with leshan in my branch, so the approach works, it's just not pretty so wondering what's the best direction here.

erkki, 2019-05-24:
.. I'm still trying to figure out the available Kconfig values .. I guess what I'm asking is clarification of how should CONFIG_NET_IPV4/CONFIG_NET_IPV6 combine with CONFIG_NET_RAW combine with CONFIG_NET_SOCKETS_OFFLOAD . In those cases code using the sockets api "should just work".

erkki, 2019-05-24:
I've rebased https://github.com/bear-metal/zephyr/tree/poc_lwm2m_net_raw against master + cleaned up a bit; If this looks sane at all happy to make a pull-request and any amendments.

jukkar, 2019-05-24:
I think CONFIG_NET_RAW is not a proper option in this case as it prevents use of upper layers of network stack. You should use CONFIG_NET_SOCKETS_OFFLOAD instead if the actual network stack is provided by some external entity. Then lwm2m would work correctly because it is running on top of socket API.

erkki, 2019-05-24:
CONFIG_NET_RAW is already used by the vendor, see https://github.com/NordicPlayground/fw-nrfconnect-nrf/blob/master/lib/bsdlib/Kconfig#L12 . 

CONFIG_NET_SOCKETS_OFFLOAD is also already used: https://github.com/NordicPlayground/fw-nrfconnect-nrf/blob/master/lib/bsdlib/nrf91_sockets.c#L29

I was just trying to see how feasible it would be to use the zephyr lwm2m engine (vs something else), and it turns out things mostly work, apart from some interpretation of how different Kconfig variables should combine.

Did you manage to check  https://github.com/bear-metal/zephyr/tree/poc_lwm2m_net_raw ?. 

The actual integration uses:
CONFIG_NETWORKING=y
CONFIG_NET_SOCKETS_OFFLOAD=y
CONFIG_NET_SOCKETS=y
CONFIG_NET_CONFIG_NEED_IPV4=y
CONFIG_NET_CONFIG_NEED_IPV6=y
CONFIG_BSD_LIBRARY=y

jtguggedal, 2019-05-27:
cc @rlubos

rlubos, 2019-05-27:
Hi @erkki, FYI we are dropping the `NET_RAW_MODE` usage with bsdlib in https://github.com/NordicPlayground/fw-nrfconnect-nrf. It was originally added to save some memory, but just as @jukkar mentioned, it turned out it does not fit the particular use case and brought up unnecessary limitations.

erkki, 2019-05-27:
That is good to know. Thanks @rlubos and everyone else.

I believe we would still need https://github.com/bear-metal/zephyr/commit/6fbefc15f921eea4726c2d434f39a310af8036d7#diff-fe5ac3fd49ff5b4af7063ea2ba575077R201 and https://github.com/bear-metal/zephyr/commit/6fbefc15f921eea4726c2d434f39a310af8036d7#diff-fe5ac3fd49ff5b4af7063ea2ba575077R247  or define the CONFIG_NET_*_COUNT in bsdlib internally. 

Closing the issue here for now as it's dependent on Nordic's followup work.

closed 2019-05-27 by erkki