zephyrproject-rtos/zephyr #16371: Settings: should leave important decisions to the application

rakons, 2019-05-24:
**Is your enhancement proposal related to a problem? Please describe.**
Settings module does some arbitrary decisions thus it can lead to system failure or data loss.
Settings did 2 things that it really should not be allowed to do without the application interaction:

1. Clears flash if it cannot initialize: This may lead to data loss. Imagine that you have upgraded the application and the data format changed. In some application, it does not matter and you can just clear the data. But in some others, we cannot afford configuration lost. It should be an application decision to clear the flash and try the settings initialisation one more time. Some applications would show a warning then to the user and allow him to downgrade to restore the settings.

2. It crashes the whole system if it cannot initialize. What? How important is ```settings``` module? One more time - this should be the application decision. The way it works right now, it may lead to the whole system crash just because settings module is broken. What if settings module is implemented on file backend and uses FLASH card? If settings module would not use kernel panic inside the application may at least display some notification like: "Settings file not present, do you wish to create empty one?" or even "Insert SD Card". In the worst case, in many... many cases application may still continue to work even if the settings module is non-functional.

**Describe the solution you'd like**
1. Settings module should allow the application to decide if it prefers to erase the backend memory.
2. Settings module should return error code not crash the whole system.

**Describe alternatives you've considered**
NULL

rakons, 2019-05-24:
cc @nvlsianpu @carlescufi

rakons, 2019-05-24:
This also relates partially to the #16275

nvlsianpu, 2019-05-24:
1. yes: Settings module should allow the application to decide if it prefers to erase the backend memory. Should be able to inform the application about unknow format which open possibility to fix or run an adaptation patch.
2. yes: because of similar reasons.