zephyrproject-rtos/zephyr #16375: net: ipv4: udp: Zephyr does not reply to a valid datagram with checksum zero

rmstoi, 2019-05-24:
If Zephyr receives datagram with valid UDP checksum zero, it does not reply and drops the packet. Instead, a reply to this packet must be sent. Attached also Linux pcap for this case.

[udp-checksum-zero-linux.pcap.gz](https://github.com/zephyrproject-rtos/zephyr/files/3216312/udp-checksum-zero-linux.pcap.gz)
[udp-checksum-zero-zephyr.pcap.gz](https://github.com/zephyrproject-rtos/zephyr/files/3216313/udp-checksum-zero-zephyr.pcap.gz)

tbursztyka, 2019-05-24:
A checksum field full of zeros in not a valid checksum. It means "no checksum".
See RFC 768 

> An all zero  transmitted checksum  value means that the transmitter  generated  no checksum  (for debugging or for higher level protocols that don't care).

So I guess if that field is full of 0s we should not check it at all.

jukkar, 2019-05-24:
> So I guess if that field is full of 0s we should not check it at all.

That is probably the proper thing to do. I just wonder if this behavior should be configurable via Kconfig option.

tbursztyka, 2019-05-24:
Good question indeed.

commit 274db4f3 (2019-05-27): net/udp: Accept UDP packet with missing checksum

Having a checksum of zeros in UDP means "missing checksum" and is a
valid case as per RFC 768:

"An all zero transmitted checksum value means that the transmitter
generated no checksum (for debugging or for higher level protocols
that don't care)."

Such support is made possible by adding a new Kconfig option named
CONFIG_NET_UDP_MISSING_CHECKSUM.

However, that is valid only for IPv4. For IPv6, see the RFC 2460
section 8.1:

"Unlike IPv4, when UDP packets are originated by an IPv6 node, the UDP
checksum is not optional."

So the UDP checksum w [... cut]
PR #17914 (2019-08-15): [Backport v1.14-branch] Networking: fix UDP checksum related issues
Backport ~~#16347~~ #16409

Fixes #16379
Fixes #16375
commit 7ddd15c2 (2019-08-15): net/udp: Accept UDP packet with missing checksum

Having a checksum of zeros in UDP means "missing checksum" and is a
valid case as per RFC 768:

"An all zero transmitted checksum value means that the transmitter
generated no checksum (for debugging or for higher level protocols
that don't care)."

Such support is made possible by adding a new Kconfig option named
CONFIG_NET_UDP_MISSING_CHECKSUM.

However, that is valid only for IPv4. For IPv6, see the RFC 2460
section 8.1:

"Unlike IPv4, when UDP packets are originated by an IPv6 node, the UDP
checksum is not optional."

So the UDP checksum w [... cut]
closed 2019-05-27 by jukkar