zephyrproject-rtos/zephyr #16379: net: ipv6: udp: Zephyr replies with illegal UDP checksum zero

rmstoi, 2019-05-24:
If Zephyr receives IPv6 UDP datagram with valid UDP checksum 0xffff, it replies with IPv6/UDP having illegal checksum zero. Zephyr replies correctly to the same UDP datagram in IPv4. Attached also Linux pcap for this case.

[zephyr-udpv4-checksum-good.gz](https://github.com/zephyrproject-rtos/zephyr/files/3216825/zephyr-udpv4-checksum-good.gz)
[zephyr-udpv6-checksum-bad.gz](https://github.com/zephyrproject-rtos/zephyr/files/3216826/zephyr-udpv6-checksum-bad.gz)
[linux-udpv6-checksum-good.gz](https://github.com/zephyrproject-rtos/zephyr/files/3216827/linux-udpv6-checksum-good.gz)

rmstoi, 2019-05-24:
[linux-udpv4-checksum-good.gz](https://github.com/zephyrproject-rtos/zephyr/files/3216866/linux-udpv4-checksum-good.gz)

rmstoi, 2019-05-24:
It seems that Linux sets the checksums in both cases, IPv4 and IPv6, to 0xffff_

tbursztyka, 2019-05-24:
In this case as well as in #16375, I think it is the same root cause in UDP side.

commit 77062582 (2019-05-27): net/udp: Computed checksum field should not be 0

From RFC 768, in "Fields":
"If the computed  checksum  is zero,  it is transmitted  as all ones"

Fixes #16379
PR #17914 (2019-08-15): [Backport v1.14-branch] Networking: fix UDP checksum related issues
Backport ~~#16347~~ #16409

Fixes #16379
Fixes #16375
commit caacf7ca (2019-08-15): net/udp: Computed checksum field should not be 0

From RFC 768, in "Fields":
"If the computed  checksum  is zero,  it is transmitted  as all ones"

Fixes #16379
closed 2019-05-27 by jukkar