zephyrproject-rtos/zephyr #16387: STM32wb55 bluetooth samples fail

yhvh, 2019-05-24:
Using a early version of the STM32wb55 most of the Bluetooth samples fail to connect with either Android app or bluetoothctl. As mentioned in 
#14188 this is possibly because of a need to upgrade the M0 core to one which works with zephyr. How do I go about doing that?
Ping @erwango

Steps to reproduce the behavior:
1. Build one of samples/Bluetooth/peripheral*
2. Flash to early version STM32wb55
3. Attempt to connect via bluetoothctl
4. Observe lack of connection

**Expected behavior**
The connection should succeed 

**Impact**
Unable to use #14188

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK
 - #14188

**Additional context**
STM32wb55 early release version

carlescufi, 2019-05-24:
I understand from @erwango that the early versions of the devkit are not supported in Zephyr

erwango, 2019-05-27:
Reopening
Early sample versions are indeed not supported for now, but it makes sense to check what it costs to get them working. 

@yhvh, as a start: can you try to get a STM32WB stm32cube package from st.com and follow the steps indicated to update M0 f/w (in Projects/STM32WB_Copro_Wireless_Binaries/Release_Notes.html, using stm32wb5x_BLE_Stack_fw.bin)?

yhvh, 2019-05-27:
@erwango This worked, following the instructions for the f/w version on the board

- Upgraded FUS, then the BLE stack.
- Tested the CRC and HR peripherals on top of #14188 with nRF android application
- Observed notification stream

Thanks!

erwango, 2019-05-27:
Closing (and keeping as reference for future similar question).

closed 2019-05-27 by erwango

pavel-demin, 2020-09-12:
I have the same problem with P-NUCLEO-WB55 EPNWB55$CU4. STM32WB55 on this board doesn't have FUS and newer versions of the M0 encrypted firmware can't be installed on this board. Would it be possible to make this kind of board working with  Zephyr's BLE stack?

The problem is the same with the peripheral* and st_ble_sensor samples. The host stops communicating with the controller right after connection, possibly due to an unhandled event:
```
[00:00:06.586,000] <inf> hci_ipm: EVT: hcievt->evtserial.evt.evtcode: 0x3e
[00:00:06.586,000] <inf> main: Connected
[00:00:06.587,000] <inf> hci_ipm: CMD: code 0x2016 plen 2
[00:00:06.587,000] <inf> hci_ipm: EVT: hcievt->evtserial.evt.evtcode: 0x0f
[00:00:06.782,000] <inf> hci_ipm: EVT: hcievt->evtserial.evt.evtcode: 0x01
[00:00:06.782,000] <wrn> bt_hci_core: Unhandled event 0x01 len 40: 07000300040002b9000000000000000~

[00:00:36.816,000] <inf> hci_ipm: EVT: hcievt->evtserial.evt.evtcode: 0x05
[00:00:36.817,000] <inf> main: Disconnected (reason 19)
[00:00:36.817,000] <inf> hci_ipm: CMD: code 0x200a plen 1
[00:00:36.817,000] <inf> hci_ipm: EVT: hcievt->evtserial.evt.evtcode: 0x0e
```

It seems that something is wrong with the HCI_COMMAND_STATUS_EVENT response to the HCI_LE_READ_REMOTE_FEATURES command.

erwango, 2020-09-14:
@pavel-demin I'm truly sorry but it is not possible to provide support on these samples since the binary is quite specific.

athulksriod, 2024-12-02:
Hi all ,  this is i got when running the stm32wb_ble sample code                                                                                 
*** BStarting Beacon Demo
                                                     
[00:00:03.127,000] [1;31m<err> os: ***** USAGE FAULT *****[0m
               
[00:00:03.127,000] [1;31m<err> os:   Illegal use of the EPSR[0m
             
[00:00:03.127,000] [1;31m<err> os: r0/a1:  0x0800c69c  r1/a2:  0x20003108  r2/a
3:  0x00080000[0m
                                                            
[00:00:03.127,000] [1;31m<err> os: r3/a4:  0x00000000 r12/ip:  0x200002e0 r14/l
r:  0x08005b21[0m
                                                            
[00:00:03.127,000] [1;31m<err> os:  xpsr:  0x20000000[0m
                    
[00:00:03.127,000] [1;31m<err> os: Faulting instruction address (r15/pc): 0x000
00000[0m
                                                                     
[00:00:03.127,000] [1;31m<err> os: >>> ZEPHYR FATAL ERROR 35: Unknown error on 
CPU 0[0m
                                                                     
[00:00:03.127,000] [1;31m<err> os: Current thread: 0x20000a80 (unknown)[0m
  
[00:00:03.186,000] [1;31m<err> os: Halting system[0m
                        
                                                                                
                                       
i update the m0 core with stm32wb5x_BLE_HCILayer_fw V1.15.0  but the code getting stuck in FUNC_NORETURN __weak void arch_system_halt(unsigned int reason) function the reason is 35

erwango, 2024-12-02:
@athulksriod Please raise a new issue and provide all required information to assist in debug (Zephyr version used, sample in use, ..)

athulksriod, 2024-12-02:
zephyr version :3.7.0

sample : zephyr\boards\stm32\power_mgmt\stm32wb_ble

board : nucleo-wb55rg

Hi all , this is i got when running the stm32wb_ble sample code #zephyr #stm32wb55 #ble
*** BStarting Beacon Demo

[00:00:03.127,000] �[1;31m os: ***** USAGE FAULT *****�[0m

[00:00:03.127,000] �[1;31m os: Illegal use of the EPSR�[0m

[00:00:03.127,000] �[1;31m os: r0/a1: 0x0800c69c r1/a2: 0x20003108 r2/a
3: 0x00080000�[0m

[00:00:03.127,000] �[1;31m os: r3/a4: 0x00000000 r12/ip: 0x200002e0 r14/l
r: 0x08005b21�[0m

[00:00:03.127,000] �[1;31m os: xpsr: 0x20000000�[0m

[00:00:03.127,000] �[1;31m os: Faulting instruction address (r15/pc): 0x000
00000�[0m

[00:00:03.127,000] �[1;31m os: >>> ZEPHYR FATAL ERROR 35: Unknown error on
CPU 0�[0m

[00:00:03.127,000] �[1;31m os: Current thread: 0x20000a80 (unknown)�[0m

[00:00:03.186,000] �[1;31m os: Halting system�[0m

i update the m0 core with stm32wb5x_BLE_HCILayer_fw V1.15.0 but the code getting stuck in FUNC_NORETURN __weak void arch_system_halt(unsigned int reason) function the reason is 35

/* LCOV_EXCL_START */
FUNC_NORETURN __weak void arch_system_halt(unsigned int reason)
{
ARG_UNUSED(reason);

/* TODO: What's the best way to totally halt the system if SMP

is enabled?
*/
(void)arch_irq_lock();
for (;;) {
/* Spin endlessly */
}
}

in this infinite for loop .

@erwango
https://github.com/zephyrproject-rtos/zephyr/issues/82394