zephyrproject-rtos/zephyr #16406: west: runners: Add --id and --chiperase options

carlescufi, 2019-05-27:
Add generic options for identifying a board and erasing the chip so that users can invoke them, on runners that support them, without having to know the underlying option of the tool used.

```
west flash --id 00234567 --chiperase
```

mbolivar-nordic, 2022-05-25:
We now have --dev-id and --erase options, which individual runners can choose to support or not, but which are standardized across all runners using the RunnerCaps mechanism:

https://github.com/zephyrproject-rtos/zephyr/blob/25e6279493d673ec76ef5cb72bbfb0b651274144/scripts/west_commands/runners/core.py#L199

closed 2022-05-25 by mbolivar-nordic

boaks, 2024-08-10:
In my experience there is a difference in using "nrfjprog" with "--chiperase" or "--sectorerase". Using "west flash" results in "--sectorerase" and using "-O--chiperase"  in an error.

```
FATAL ERROR: command exited with status 13: nrfjprog --program build_nrf9161dk_nrf9161_ns/zephyr/merged.hex --sectorerase --verify -f NRF91 --snr 1050907573 --chiperase
```

Some cases requires the "--chiperase" and so that will not work with "west flash".