zephyrproject-rtos/zephyr #16426: Missing included dependencies in many header files

greglund, 2019-05-27:
**Is your enhancement proposal related to a problem? Please describe.**
A clear and concise description of what the problem is.
I'm new to Zephyr, but I'm a fairly seasoned developer on bare metal systems, and C/C++.  I was integrating a piece of software into a zephyr application and was then told by my compiler that I was missing a reference to 'size_t' and that I probably forgot to include stddef.h.  The offending file was actually bluetooth/crypto.h in the Zephyr tree.  Now I could easily work around this and include stddef.h myself before including bluetooth/crypto.h, but I strongly believe that if a header file is making use of an api, it should include the file itself for completeness, and to keep me from having to include that file out of order in my own list of dependencies.  

I thought that maybe it was just a small instance affecting just this one header file, but it is endemic to many headers I sampled after this.  In my opinion, any header provided by Zephyr should completely include its own dependencies so that developers using those headers only need to include that header.  
#1, it is self-documenting and obvious
#2, it allows a developer to use small parts of the API without messing around to figure out all the other parts of the API depend upon
#3, it is polite, and allows the developer to include those dependencies in the order of their choice (alphabetical is my choice, and this breaks that choice and forces me to figure out the chain of includes required to ensure dependencies are included before each file needs them)

I cannot think of any downsides to including all the dependencies, and I have only come across the lack of includes in rare instances where the developer of the API is new/inexperienced, or it's a commercial environment promoting vendor lock-in.  

I don't see any discussion around it in any open/closed issues, which surprises me a little. I would like to open the floor to discussion about this to find out if anyone else feels the same, or if there's a valid explanation as to why I'm the only one that feels this way!

Cheers,

Greg

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or graphics (drag-and-drop an image) about the feature request here.

aescolar, 2019-05-28:
CC: @jhedberg @galak @nashif

jhedberg, 2019-05-28:
I fully agree with this - all public header files should be self-sufficient. @greglund since you already have a list of (some) offending files and what dependencies they're missing, maybe you could create an initial pull request for this?

SebastianBoe, 2019-08-13:
Yes, we want this.

But I don't see the value in having an open issue tracking this, because even if we were able to achieve this for all headers, we wouldn't be able to prove it, or to enforce that all headers would stay compliant.

So I'm closing this.

Thank you for the good description of why includes should be organized like this, it will prove useful in code review.

closed 2019-08-13 by SebastianBoe