zephyrproject-rtos/zephyr #16428: samples/gui/lvgl does not work on PCA10056

pabigot, 2019-05-27:
At a minimum the device configuration is confused: the PCA10056 pins in the documentation are for SPI_0, but the code tries to work with SPI_1. The sample was originally developed for the nRF52840 Preview DK, and long before PCA10056 DTS was set up, so maybe that explains it.

After correcting that, and putting a logic analyzer on to confirm that the pins and SPI bus transactions appear sane, the display remains blank (it turns on and off, but there's nothing but greyscreen on the display).

The initialization sequence doesn't match that of the [Adafruit library](https://github.com/adafruit/Adafruit_ILI9340); some of that is probably because their library uses 16-bit color while Zephyr's is using 18-bit.  The differences are too great to isolate why it doesn't work.

When I go to the [product page](https://www.adafruit.com/product/1480) Adafruit tells me I bought the display in late 2014, so maybe it's a hardware version compatibility problem: mine probably has an ILI9340 while the newer models are ILI9341.

But I suspect it's bit rot.

jfischer-phytec-iot, 2019-05-27:
Please take a look at #13980. There is an other sample `samples/display/ili9340` and the #13980 fixes a few similar issues.

pabigot, 2019-05-27:
@jfischer-phytec-iot Wow.  OK, that's still screwed up, and may not help me with lvgl, but I'll try to figure it out and post comments there.

pabigot, 2019-05-28:
Things are a bit better with #13980 applied, but it still doesn't work.

I used the nrf52840_pca10056.overlay from #13980, but removing the non-standard override of the SPI pins, a configuration known to work with the new display/ili9340 sample.

With the default `CONFIG_ILI9340_RGB888` and the sample original `CONFIG_LVGL_BITS_PER_PIXEL=24` what I get on the display is strips of white and black, with a few black bits in the white band that are misplaced content from the periodic display of the counter.

So `CONFIG_LVGL_BITS_PER_PIXEL=24` is wrong.  Removing it from `nrf52840_pca10056.conf` I get a white background with the tex from the periodic display smeared as a band in the middle (and a little at the bottom), as if the pixels being set are being scaled to a multiple of the correct coordinate.

If I use `CONFIG_ILI9340_RGB565=y` instead I get the same white background and the smeared bits are in slightly different positions.

This could be a problem with the lvgl layer, or it could be a problem in #13980 that isn't exposed by drawing colored squares in the corners.

vanwinkeljan, 2019-05-28:
@pabigot In case of ```CONFIG_ILI9340_RGB888``` the lvgl pixel format is RGBA and ```CONFIG_LVGL_BITS_PER_PIXEL=24``` needs to be set else you will transfer 32-bits per pixel instead of 24-bit.

I will try to have a look next week.

pabigot, 2019-05-28:
@vanwinkeljan Thank you.  With a hint from @leewkb4567 I've got this working with `CONFIG_ILI9340_RGB565`  (also needs `CONFIG_LVLG_COLOR_DEPTH=16`), but the default `RGB888` configuration still produces strips of white and black bars.

vanwinkeljan, 2019-05-28:
RGB888 problem is reproducible with native_posix board so it is not ILI9340 display driver related

PR #16458 (2019-05-30): Corrected RGB888 buffer pointer in LVGL write function
Corrected buffer pointer in LVGL zephyr_vdb_write function for
converting RGBA8888 to RGB888 pixel format.

Fixes #16428
closed 2019-05-30 by nashif