zephyrproject-rtos/zephyr #16432: Weird link error of the console sample!

madandyliu, 2019-05-28:
**Describe the bug**

The console getline sample works well if I change nothing. But once I modify the linking library sequence in the last step produced by ninja. It still links successful, but the program would fail when executing.

I reproduced this bug on both the qemu_cortex_m3 and my mimxrt1050_evk platform. With both Zephyr SDK 0.10.0 and gcc-arm-none-eabi-7-2018-q2-update toolchains.

**To Reproduce with Zephyr SDK 0.10.0 for qemu_cortex_m3**

1. `cd samples/subsys/console/getline`
2. `mkdir build && cd build`
3. `cmake -DBOARD=qemu_cortex_m3 ..`
4. `ninja -v` (now you can do `ninja run`, and the program works well)
5. Copy the last whole linking command, the original linking sequence after `-Wl,--whole-archive` is:
```
-Wl,--whole-archive
app/libapp.a
zephyr/libzephyr.a
zephyr/arch/arch/arm/core/libarch__arm__core.a
zephyr/arch/arch/arm/core/cortex_m/libarch__arm__core__cortex_m.a
zephyr/lib/libc/minimal/liblib__libc__minimal.a
zephyr/drivers/gpio/libdrivers__gpio.a
zephyr/drivers/serial/libdrivers__serial.a
-Wl,--no-whole-archive
```
6. Change it into the following sequence:
```
-Wl,--whole-archive
app/libapp.a
zephyr/arch/arch/arm/core/libarch__arm__core.a
zephyr/arch/arch/arm/core/cortex_m/libarch__arm__core__cortex_m.a
zephyr/lib/libc/minimal/liblib__libc__minimal.a
zephyr/drivers/gpio/libdrivers__gpio.a
zephyr/drivers/serial/libdrivers__serial.a
zephyr/libzephyr.a
-Wl,--no-whole-archive
```
7. Rerun the changed linking command.
8. `ninja run` in the build directory, fail would happen once you press any key:
```
***** Booting Zephyr OS 1.14.0 *****
Enter a line
***** USAGE FAULT *****
  Illegal use of the EPSR
***** Hardware exception *****
Current thread ID = 0x20000184
Faulting instruction address = 0x0
Fatal fault in essential thread! Spinning...
```

**How did this happen**

I made my own script to search and link different libraries. I found that once I use the console module provided by zephyr, it would crash the program. Finally I found the linking sequence would affect the result. The strange thing is that the compiler wouldn't give any warning but just fail when executing the result.

The following command is the original whole linking command on my computer:

```
/zephyr-sdk/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc    zephyr/CMakeFiles/kernel_elf.dir/misc/empty_file.c.obj zephyr/CMakeFiles/kernel_elf.dir/isr_tables.c.obj  -o zephyr/zephyr.elf  -T zephyr/linker_pass_final.cmd -Wl,-Map=/home/andy/oriZephyr/zephyr-v1.14.0/samples/subsys/console/getline/build/zephyr/zephyr.map -u_OffsetAbsSyms -u_ConfigAbsSyms -Wl,--whole-archive app/libapp.a zephyr/libzephyr.a zephyr/arch/arch/arm/core/libarch__arm__core.a zephyr/arch/arch/arm/core/cortex_m/libarch__arm__core__cortex_m.a zephyr/lib/libc/minimal/liblib__libc__minimal.a zephyr/drivers/gpio/libdrivers__gpio.a zephyr/drivers/serial/libdrivers__serial.a -Wl,--no-whole-archive zephyr/kernel/libkernel.a zephyr/CMakeFiles/offsets.dir/arch/arm/core/offsets/offsets.c.obj -L"/zephyr-sdk/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/8.3.0/thumb/v7-m/nofp" -L/home/andy/oriZephyr/zephyr-v1.14.0/samples/subsys/console/getline/build/zephyr -lgcc -mthumb -mcpu=cortex-m3 -nostdlib -static -no-pie -Wl,-X -Wl,-N -Wl,--gc-sections -Wl,--build-id=none -Wl,--sort-common=descending -Wl,--sort-section=alignment -Wl,--orphan-handling=warn -mabi=aapcs -march=armv7-m && cd /home/andy/oriZephyr/zephyr-v1.14.0/samples/subsys/console/getline/build/zephyr && /usr/bin/python3 /home/andy/oriZephyr/zephyr-v1.14.0/scripts/check_link_map.py zephyr.map && /zephyr-sdk/arm-zephyr-eabi/bin/arm-zephyr-eabi-objdump -S zephyr.elf > zephyr.lst && /zephyr-sdk/arm-zephyr-eabi/bin/arm-zephyr-eabi-readelf -e zephyr.elf > zephyr.stat
```

**Environment**

 - OS: Ubuntu 18.04
 - Toolchain: Both Zephyr SDK 0.10.0 and gcc-arm-none-eabi-7-2018-q2-update
 - Version used: Zephyr 1.14.0 Release

aescolar, 2019-05-28:
CC: @SebastianBoe

SebastianBoe, 2019-05-28:
Zephyr links in two passes and requires the two passes to place the same code/data at the same place.

The order in which libraries are given to the linker affects where the linker puts them.

Could it be that the requirement of things being placed the same place in both links is violated, and thereby corrupting the build?

Try reproducing this with a build that only does a single linker pass.

SebastianBoe, 2019-05-28:
I want to classify this as a question until the root cause is found, it is not supported to copy GCC/LD commands and move stuff around.

ioannisg, 2019-05-28:
@SebastianBoe could this Question, now, be closed?

madandyliu, 2019-05-29:
@SebastianBoe Yes, it's the problem.

The first linking pass generates `zephyr_prebuild.elf`, and a python script would generate the `isr_tables.c` based on it. Different library sequence would affect the generated `isr_tables.c`. So the library sequence in the second linking pass have to be exact the same with the first one.

Thank you for your explanation. It had confused me for a full day : )

closed 2019-05-29 by madandyliu