zephyrproject-rtos/zephyr #16436: Organize generated include files

erwango, 2019-05-28:
**Is your enhancement proposal related to a problem? Please describe.**
We can generate a lot of include files at build, all under `zephyr/include/generated`.
We're generating files that belong to various type: header, linker files, ...

It would be nice to have a sublevel folder hierarchy.
Note: there is already a sublevel `zephyr/include/generated\syscalls`

**Describe the solution you'd like**
Generate include files under a sublevel category:
```
generated
 └── include
     ├── linker
     ├── dts
     ├── kernel
     ├── ...
     └── syscalls
```

**Describe alternatives you've considered**
Sit down and watch the mess growing

ulfalizer, 2019-05-28:
The naming seems a bit redundant at the moment, like `zephyr/include/generated/generated_dts_board_fixups.h`. I wonder if it could just be `zephyr/include/dts_board_fixups.h`, since it's easy to understand that it's generated anyway when it's in the build directory.

Flat hierarchies are nice, IMO, and avoiding directories that only have another directory in them (like `zephyr/include/`, which only seems to have `zephyr/include/generated` in it).

erwango, 2019-05-29:
> The naming seems a bit redundant at the moment, like zephyr/include/generated/generated_dts_board_fixups.h. I wonder if it could just be zephyr/include/dts_board_fixups.h, since it's easy to understand that it's generated anyway when it's in the build directory.

What about `zephyr/include/dts/board_fixups.h` then ? ;-)

> Flat hierarchies are nice, IMO, and avoiding directories that only have another directory in them (like zephyr/include/, which only seems to have zephyr/include/generated/ in it).

Agreed, not sure the `include` makes any sense here (besides we can find .c and .ld files in it).

erwango, 2020-05-06:
Closing as the current solution is far from description on this PR

closed 2020-05-06 by erwango