zephyrproject-rtos/zephyr #16438: fs/FCB fs/NVS : requires unaligned flash read-out length capabilities,

nvlsianpu, 2019-05-28:
FCB and NVS is implemented in such a way that it ensure that flash API read call is always aligned in read-out offset while read-out len can be unaligned.

So fare this haven't been causing any issue to real hardware, but issue appear on flash drive simulation while check for alignment of read calls was enabled.

Discussion Sniped from #16405 where issue was raised:

> No it is not. Line 32 in fcb_elem_info.c:

> rc = fcb_flash_read(fcb, loc->fe_sector, loc->fe_elem_off, tmp_str, 2);
The position is aligned. The size is not.
Line 79 in the same file:

> u8_t fl_crc8;
...
rc = fcb_flash_read(fcb, loc->fe_sector, off, &fl_crc8, sizeof(fl_crc8));
The fcb.c file, line 266:

>rc = fcb_flash_read(fcb, sector, 0, fdap, sizeof(*fdap));
Sure, it is aligned to 8... but there is no control about if it is aligned to FLASH_SIMULATOR_PROG_UNIT.

> I do not even want to search more. If memory requires aligned read access this can always be fixed in the driver layer to allow to read unaligned data. The write is problematic. When reading you can always throw away some data.

### Prefer solution:
Align flash API read implementation in such a way that each of them will suport at last unaligned read-out length. It appear that unaligned read acces is `cheap` in sense of software implementation.

Current state of flash drivers is (pleas correct <b>here</b> if I missed something):
QMSI has requirement for 4 B alignment of offset and length.
NIOS2 has requirement for 4 B alignment of offset.
NRF, MCUX, Geco, STM, SAM, SPI-NOR, w25qxxdv are capable of unaligned readout

### alternative solution
Patch both FCB and NVS in order to support flash driver which requires aligned read-ot size.

Laczen, 2019-05-28:
@nvlsianpu, I propose a different solution: During the initialisation of fcb/nvs a check is performed on the flash requirements (i.e. for nvs try a read with unaligned length), if this check fails the initialisation fails with error code -ENXIO or -ENOTSUP.

Whatever flash device is added later on, and whatever it's possibilities are, it is clear that nvs/fcb will not work.

Laczen, 2019-05-28:
@nvlsianpu, included is a patch for nvs that checks for the unaligned read length size and also adds a check if the filesystem offset is aligned.  As you know I am no longer creating PR's. This will however make the nvs test fail with standard simulator settings. The nvs test must have been very lucky not to do any unaligned read.

[nvs_patch.txt](https://github.com/zephyrproject-rtos/zephyr/files/3228045/nvs_patch.txt)

nvlsianpu, 2020-11-13:
Allow unaligned read is flash_read API implementation requirement since a while.

closed 2020-11-13 by nvlsianpu