zephyrproject-rtos/zephyr #16439: driver/flash: unify read alignment requirements

nvlsianpu, 2019-05-28:
### Problem to be covered:
Flash read API implementation have different requirement regards read-out offset and read-out length. These requirements are not precised by any API description.

Most of drivers allow unaligned read,. One driver enforce aligned access in terms on offset and length. One driver requires aligned offset only.

Problem with this diverse behavior cause unwanted complication in upper layer in order to adapt to what any flash driver supports. It cause for instance issue described here: #16438

### Solution:
Sufficient: At last make each driver capable to read unaligned length of data. Better if unaligned offset will be ensured for any driver.

carlescufi, 2019-06-04:
Conclusion: `flash_read()` users should be able to use any address and length, aligned or unaligned.

@nashif can you see if the owners of QMSI and NIOS2 could add software emulation for unaligned reads?

nvlsianpu, 2019-06-04:
@ramakrishnapallala  Are you able to align NIOS2 driver? Saw that you contribute it.

nvlsianpu, 2019-07-12:
> @nashif  can you see if the owners of QMSI and NIOS2 could add software emulation for unaligned reads?
@nasthif ping
I want to act on API requirement change in order to solve this issue.

d3zd3z, 2019-12-03:
Unlike writes, it should always be safe for the driver to read more data is then only return the data the user has requested. I think this change is a good idea.

nvlsianpu, 2020-03-20:
QMSI driver was droped.
NIOS2 driver haw been improved

d3zd3z, 2020-05-22:
> Unlike writes, it should always be safe for the driver to read more data is then only return the data the user has requested.

I should clarify this a bit, because it isn't quite true. It should be safe to read more data, up to the size of the read alignment. For example, a device that has 32-bit internal reads, and uses an ECC to protect those will always be safe to read a full 32-bits. But, if the driver were to read 64-bits from this device, it could receive an error on the second 32-bits if it hadn't been written.

Devices do not behave the same as far as reads from unwritten flash. Most seem to return a consistent pattern (all 1s or all 0s), but at least one, the LPC55S69, will bus-fault. #25206 attempts to make read more consistent on this device, but it will still return `-ENODATA` on unwritten data, or just `-EIO` if it isn't erased (which it would do for a larger read). This device has a write-granularity of 512, so 32-bit reads should always be safe.

So, I think this is safe to round the reads up, but we have to be careful about the size of the reads we do, and that we don't make them larger than the underlying write granularity of the device.

PR #23628 (2020-08-05): include/drivers/flash: document on the unrestricted alignment of reads
Unaligned read-out capability become fact among all drivers.
Let's cut this in stone as API requirement.

fixes #16439
commit 1d8cf859 (2020-08-05): include/drivers/flash: document on the unrestricted alignment of reads

Unaligned read-out capability become fact among all drivers.
Let's cut this in stone as API requirement.

fixes #16439
closed 2020-08-05 by nashif