zephyrproject-rtos/zephyr #16444: drivers/flash/flash_simulator: Support for required read alignment

rakons, 2019-05-28:
**Is your enhancement proposal related to a problem? Please describe.**
As it was suggested in #16405 PR, we should be able to simulate specific FLASH memory that has a requirement to read alignment in start address and the size separately.

@Laczen:
> to make the flash_simulator capable of simulating memories that have requirements on the alignment it might be good to create 2 defines: CONFIG_FLASH_SIMULATOR_ALIGNED_READ_START and CONFIG_FLASH_SIMULATOR_ALIGNED_READ_LENGTH, with the second automatically selecting the first.

I also think that such simulator to be complete should have separate settings for the required READ alignment, something like: ```FLASH_SIMULATOR_PROG_UNIT``` -> ```FLASH_SIMULATOR_READ_UNIT``` for the read function checking.

**Describe the solution you'd like**
Additional configuration for aligned read requirements.

**Describe alternatives you've considered**
Not implementing it? Currently, we have qemu running as expected, but we cannot simulate every case on flashsimulator.

**Additional context**
Currently, we can only turn reading alignment checking on and off. It is also tightly coupled with the writing alignment.

Laczen, 2019-05-28:
I created a new issue while you where making this one.

rakons, 2019-05-28:
Could you add to the #16446 the idea to separate required alignment values between reading and writing. I can image a memory that requires, for example, an alignment of 8 to write and 4 to read. Also please use ```#``` at the beginning of PR numbers and issues numbers to allow github engine to link them correctly.

Then this issue may be closed as duplicated.

d3zd3z, 2019-12-03:
If #16439 is implemented, then this issue would not be relevant.

henrikbrixandersen, 2022-11-17:
@rakons, @d3zd3z Seeing that https://github.com/zephyrproject-rtos/zephyr/issues/16439 was implemented and merged - can this be closed?