zephyrproject-rtos/zephyr #16452: drivers: ethernet: stm32, sam, mcux

alexanderwachter, 2019-05-28:
**Describe the bug**
Random generated MAC addresses flip the wrong bit for LAA.
The LAA bit is the second bit in the first OUI byte, not the last byte in the NIC specific part.
Generally all assigned addresses should have the LAA bit set when using the vendo's OUI not only random generated.

**Impact**
MAC address looks like it is UAA address but is LAA.

pfalcon, 2019-05-28:
@alexanderwachter: Can you please update the title of this ticket to correspond to the bug reported?

ioannisg, 2019-08-27:
@pfalcon is this issue being looked at?
I am checking if this could be done for 2.0.

alexanderwachter, 2019-08-27:
@pfalcon No, it does not break anything. It just violates the IANA specifications.
Assumed I'm not wrong here and the actual implementation is correct.

alexanderwachter, 2019-08-27:
@pfalcon If you can confirm that the actual implementation is wrong, I can fix the drivers. But IMO the priority is low.

pfalcon, 2019-08-27:
I don't look into this so far. Is this issue breaking something? I didn't notice Ethernet not working due to addresses being assigned. If so, I'd suggest to set priority as low.

pfalcon, 2019-08-27:
> But IMO the priority is low.

Thanks for confirming. 3 days before release, there're bigger priorities to look into, and I have backlog of tasks otherwise, but we'll get to this issue too of course.

PR #19237 (2019-09-18): drivers: ethernet: stm32, sam, mcux: correctly set LAA bit
When randomly generating MAC addresses they will always be
locally administrated addresses, so the LAA bit should be set.
The LAA bit is the 2nd bit of the 1st byte of the MAC address
not the 2nd bit of the 4th byte.

Fixes:  #16452
commit c0ae6746 (2019-09-18): drivers: ethernet: stm32, sam, mcux: correctly set LAA bit

When randomly generating MAC addresses they will always be
locally administrated addresses, so the LAA bit should be set.
The LAA bit is the 2nd bit of the 1st byte of the MAC address
not the 2nd bit of the 4th byte.

Fixes: #16452
closed 2019-09-18 by jukkar