zephyrproject-rtos/zephyr #16463: tests/subsys/settings/fcb_init fails on twrke18f board

hakehuang, 2019-05-29:
**Describe the bug**
build the tests/subsys/settings/fcb_init application, run and test_init fails

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=twr_ke18f ..
3. make
4. See error

**Expected behavior**
the setting_fcb test can PASS

**Impact**
settings fcb can not work

**Screenshots or console output**
```
***** Booting Zephyr OS 1.14.99 *****

Running test suite test_initialization

===================================================================

starting test - test_init

    Assertion failed at ./tests/subsys/settings/fcb_init/src/settings_test_fcb_init.c:61: test_init: (prev_int not equal to val32)

load value doesn't match to what was saved

FAIL - test_init

===================================================================

Test suite test_initialization failed.

===================================================================

PROJECT EXECUTION FAILED

```

**Environment (please complete the following information):**
 - OS: (e.g. Linux)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA : 59c8c9930d0b27e2a53e05cd5947cbe7f8bc800b

hakehuang, 2019-05-29:
@MaureenHelm @henrikbrixandersen

henrikbrixandersen, 2019-05-29:
***** Booting Zephyr OS zephyr-v1.14.0-1117-g93fb695244c6 *****
Running test suite test_initialization
===================================================================
starting test - test_init
PASS - test_init
===================================================================
Test suite test_initialization succeeded
===================================================================
PROJECT EXECUTION SUCCESSFUL

henrikbrixandersen, 2019-05-29:
@hakehuang It works for me on 59c8c9930d0b27e2a53e05cd5947cbe7f8bc800b.

I cannot find the commit SHA, you mention?
```
$ git show 59c8c9930d0b27e2a53e05cd5947cbe7f8bc800b
fatal: bad object 59c8c9930d0b27e2a53e05cd5947cbe7f8bc800b
```

hakehuang, 2019-05-30:
sorry, it the master branch the SHAshall be e72e457360ffa8b07f29f6a5ce496a62d7ceb253

henrikbrixandersen, 2019-05-30:
@hakehuang Have you tried erasing the entire flash before running this test? From a quick glance it seems that the test would fail the second time it is run without erasing the flash beforehand.

hakehuang, 2019-05-31:
@henrikbrixandersen , you are right, but this seems the fcb has problem, it supposes to erase the fcb area, before usage.

henrikbrixandersen, 2019-06-03:
@nvlsianpu Could you please take a look at this?

nvlsianpu, 2019-06-04:
@hakehuang - I understand than with storage area erased before you run the tests, tests passed?

henrikbrixandersen, 2019-06-04:
> @hakehuang - I understand than with storage area erased before you run the tests, tests passed?

Correct.

nvlsianpu, 2019-06-04:
I confirm - this test assumes that storage partition is clear before the test is run.
So the issue appears just because `west flash` doesn't erase more flash than needed for program the device, so doesn't erase especially storage partition.

This test can't ensurer storage clearance because it reboots the device as the part of the testing routine.
So need to find a way take storage into build product.

hakehuang, 2019-06-17:
> @hakehuang - I understand than with storage area erased before you run the tests, tests passed?

yes, it passes after erase.

@nvlsianpu

PR #18515 (2019-08-23): tests/subsys/settings/fcb_int: add storage preparation
The test assumes that storage is clear before it runs. This
might be not true as lot of devices are only partially erased by
'west flash'.

Patch introduce procedure for testing whether test runs the first
time and clear storage if so.

fixes #16463
commit 93702e2d (2019-08-23): tests/subsys/settings/fcb_int: add storage preparation

The test assumes that storage is clear before it runs. This
might be not true as lot of devices are only partially erased by
'west flash'.

Patch introduce procedure for testing whether test runs the first
time and clear storage if so.

The procedure uses mark which is stored inside SoC embedded program
flash. It will not work one devices on which read/write to it is
impossible.

fixes #16463
closed 2019-08-23 by ioannisg