zephyrproject-rtos/zephyr #16472: tinycrypt ecc-dh and ecc-dsa should not select entropy generator

Laczen, 2019-05-29:
## Introduction

Tinycrypt provides support for ecc-dh and ecc-dsa. 

### Problem description

When ecc-dh or ecc-dsa are selected the entropy driver is selected. There is no need for this, ecc-dh and ecc-dsa depend on a random generator but this is not provided by the entropy driver.

The addition of the entropy driver is just adding flash usage.

Even in the case entropy driver would provide a random generator, it still would not be required to include the driver as this is only required when key generation is done. To verify a dsa or to do a key exchange there is no need for the random generator.

Why do we want this change and what problem are we trying to address?

### Proposed change

Remove the selection of the entropy driver from tinycrypt Kconfig.

aescolar, 2019-05-29:
@lpereira Any thoughts?

alexanderwachter, 2019-06-07:
I think you need entropy for the nonce generation.

Laczen, 2019-06-07:
@alexanderwachter, thanks for the reply. In the general case you would need the entropy generator, but you need more than that, you need to provide a random function to use the entropy generator. This entropy does not provide.
And there also is the case where you would use ecc-dh and ecc-dh only for validation, then you don't need the entropy generation at all.
A solution could be that whenever an application needs to generate keys it will select the entropy subsystem and provide the required random function (using the entropy generator) so let the selection of entropy be a application choice. If the application would generate it's own random function (not wise) it would not select entropy saving ram and flash.

d3zd3z, 2019-12-03:
I believe this analysis is correct. Entropy/random is not necessary to verify signatures, and it is reasonable for an embedded device to be configured for only this situation. Both entropy and a CPRNG would be needed to sign images or generate keys. Hopefully this aspect will get cleaned up by upcoming random/entropy changes.

BTW, is this fixed by #19471.

Laczen, 2019-12-03:
Yes, this is fixed

closed 2019-12-03 by Laczen