zephyrproject-rtos/zephyr #16478: Bluetooth: Improper bonded peers handling

MarekPieta, 2019-05-29:
**Describe the bug**
Got double bonds after connection and security establishment with host. One of the bonds disappears after device reset.

**To Reproduce**
After advertising, connecting and pairing on one identity, called bt_unpair for another identity. There were two bonds present for identity where connection occurred (representing the same host - Windows 10). Host saw only one paired and connected bluetooth device. 

**Expected behavior**
After single connection and security establishment we should get one bond.
Bond should not be removed after device reset when settings are enabled.

jhedberg, 2019-05-29:
Could you provide some more information? Is either of the devices using privacy? By "identity" I suppose you're talking of the remote identity, since we also have a multi-identity feature for peripheral role (in which case identity is used to refer to the local identity).

jhedberg, 2019-05-29:
Also, what do you mean by getting "double bonds"? Two instances of a bt_keys struct?

MarekPieta, 2019-05-29:
By identity I meant local identities for peripherial role device. Devices are not using privacy.
By double bonds I meant two instances of bt_keys struct for the same connected host (still host sees one device).

jhedberg, 2019-05-29:
Thanks for the clarifications, however it's still a bit unclear how I should reproduce this. Could you perhaps provide a set of Bluetooth shell commands that will reproduce this (the shell supports using multiple identities)

MarekPieta, 2019-05-29:
Btw. Maybe problem can be related to bt_keys_get_addr implementation.
As I noticed, we create and return new key when we find empty one in array before we find the one which is already created for given identity and peer address (when we remove we may remove any key).
Is it ok ? Maybe we could change it to two loops so we will ensure that we will find key if it already exists.

jhedberg, 2019-05-29:
@MarekPieta yes that sounds reasonable. The bt_keys_get_addr() was implemented before we had unpair support, IIRC, i.e. back then it was not possible to have "gaps" in the keys array.

jhedberg, 2019-05-29:
@MarekPieta btw, instead of iterating twice, you could consider using similar "free slot" tracking as what `va_add()` in `mesh/cfg_srv.c` does: https://github.com/zephyrproject-rtos/zephyr/blob/974231ee6e29223c5498f1281523aabf8fb62a33/subsys/bluetooth/host/mesh/cfg_srv.c#L1158-L1180

PR #16490 (2019-05-30): Bluetooth: host: Fix bt_keys_get_addr()
Change fixes issue causing multiple keys creation for given address and identity.

Fixes: #16478
commit 47c12a07 (2019-05-30): Bluetooth: host: Fix bt_keys_get_addr()

Change fixes issue causing multiple keys creation for given address
and identity.

Fixes: #16478
commit 3f825a8c (2019-05-30): Bluetooth: host: Fix bt_keys_get_addr()

Change fixes issue causing multiple keys creation for given address
and identity.

Fixes: #16478
commit c8be3e81 (2019-05-30): Bluetooth: host: Fix bt_keys_get_addr()

Change fixes issue causing multiple keys creation for given address
and identity.

Fixes: #16478
closed 2019-05-30 by carlescufi