zephyrproject-rtos/zephyr #16482: mcumgr seems to compromise BT security

PhilippFinke, 2019-05-29:
## Introduction

I'm using a nrf52840 based custom peripheral in my project where I can upload FOTAs via mcumgr under Linux. My peripheral requests security (Mode 1, Level 4 with fixed passkey) after a connection request. All my peripheral's Characteristic and Descriptors are authenticated read/write only, except the mcumgr SMP BT service.

If I want to connect to the peripheral I get asked for the passkey as expected. There are three possibilities here:

- Entering the correct passkey -> peers will bond, connection fully established.
- Entering wrong passkey -> disconnect with authentication error.
- Entering nothing -> disconnect with authentication error after some amount of time.

Here are my security and pairing/bondig related configurations:

BT_SMP=y
BT_SMP_SC_ONLY=y
BT_SMP_SC_PAIR_ONLY=y
BT_FIXED_PASSKEY=y
BT_BONDABLE=y
BT_SETTINGS=y

### Problem description

When uploading images (or performing any other mcumgr operation on the peripheral) I won't get asked to enter a passkey, but I don't get a disconnect with an authentication error either. It seems that mcumgr is somehow able to break my security feature. I also tried this with peers that were unbonded (no previous pairing keys available) before.

This is a serious issue, because this compromises my security concept. How is this possible and how I can prevent this?

aescolar, 2019-05-29:
CC @carlescufi @jhedberg

carlescufi, 2019-05-29:
By default mcumgr does not require bonding or encryption. I am not sure the mcumgr code includes the necessary handlers to enable security. This should be made configurable in my opinion in any case.

carlescufi, 2019-05-29:
Also @PhilippFinke remember that the image signature will be checked by the bootloader and if the signature is not valid it will not be swapped, so a malicious attacker couldn't override the firmware.

PhilippFinke, 2019-05-29:
@carlescufi, this is true. But isn't there a possibility to download the firmware?

carlescufi, 2019-05-29:
> @carlescufi, this is true. But isn't there a possibility to download the firmware?

Yes, you will experience a Denial-of-Service type of attack: the attacker will be able to upload an image and reboot your device, but then it will revert back to the original image once the bootloader detects that the signature is incorrect. That said, I am hoping @joerchan can add a Kconfig option that enable security on this particular GATT service.

PhilippFinke, 2019-05-29:
This is not what I mean. I mean downloading the fw image from the peripheral.

This would permit an attacker to copy my fw to a cloned hardware of our product.

carlescufi, 2019-05-29:
> This is not what I mean. I mean downloading the fw image from the peripheral.
> 
> This would permit an attacker to copy my fw to a cloned hardware of our product.

Yes, but adding BLE security wouldn't help you there would it? because at the end of the day the attacker could still bond with the central/phone and then download if they have control of both ends.

PhilippFinke, 2019-05-29:
> Yes, but adding BLE security wouldn't help you there would it? because at the end of the day the attacker could still bond with the central/phone and then download if they have control of both ends.

But in this case he must know my fixed passkey to connect and pair to the device in the first place.

carlescufi, 2019-05-29:
Ah, if  you use a fixed passkey then you should be safe.

PhilippFinke, 2019-05-29:
I did some further research on this issue.

I copied the content of smp_bt.h and smp_bt.c and changed the permissions of the single characteristic in the SMP service from BT_GATT_PERM_WRITE to BT_GATT_PERM_WRITE_AUTHEN.

When I run mcumgr now, I always get an Insufficient authetication error, which makes sense. I'm running simple-agent on my Linux machine as global agent, that asks me to enter the passkey when I want to connect and pair to my peripheral. Unfortunately this is not the case when using mcumgr.

Laczen, 2019-06-03:
@PhilippFinke, have you considered using encrypted firmware? In that case the communication channels used to send the firmware can be insecure. When moving from slot 1 to slot 0 the firmware is decrypted, when moving from slot 0 to 1 the firmware is encrypted. If you download the firmware from the device you would get an encrypted firmware. Only a bootloader that can derive the encryption key is capable of decrypting the firmware, so someone trying to load your firmware to a different device would also need to have your bootloader.

PhilippFinke, 2019-06-03:
Hey @Laczen,

I'm already using encrypted FW to prevent users from uploading their own custom FWs. But does this prevent an attacker to simply download my bootloader and flash it to his own board? In this case he would be able to use my FW.

Laczen, 2019-06-03:
@PhilippFinke, you are right. But if a attacker can get your bootloader from your board also the combination of bootloader and firmware can be downloaded (and programmed on a new device). A secure communication channel for the upgrade does not solve this. The bootloader should be protected against readout and all jtag/swd communitcation interfaces should be disabled.

PhilippFinke, 2019-06-03:
Thanks for the information. I planned to apply read protection (and write protection for the bootloader) in my final field FW. But what I meant is downloading the bootloader (and/or FW) via mcumgr. Is that possible?

Laczen, 2019-06-03:
Hi @PhilippFinke, mcuboot cannot be upgraded via mcumgr.

PhilippFinke, 2019-06-03:
Yeah, I know. But still not what I meant. Let me try to explain:

1. Attacker uses mcumgr to download bootloader (from my board to PC!).
2. Attacker uses mcumgr to download FW (from my board to PC!).
3 Attacker flashes his board (clone of my official one) with the downloaded bootloader and FW.

This would allow any attacker to simply clone the HW of my board and to flash it with the bootloader and FW he downloaded from my official HW (he might bought one from official sources) via mcumgr. I'm asking this, because we had exactly this issue with a product by a company I worked in.

So, my question is: Is it possible to download the bootloader and/or the FW from the board to a PC via mcumgr to use it later on another board?

Laczen, 2019-06-03:
OK, I don't know mcumgr very well but step 1 should not be possible. mcumgr should only allow reading and writing to slot1 location, It should not be able to download from the bootloader location. In an ideal case mcumgr would even only have the information about where to store and read slot1, nothing more. I don't know if it is implemented like this in mcumgr, but it should be. I know it does not completely follow this rule to allow reading the version information, but the general rule should be that mcumgr only uses slot1 location information and that when you request a download you would only get the firmware in slot1.
If mcumgr does not follow this rule it should be easy to modify so that it behaves like this: the mcumgr routines that are used by the image in slot0 (the running image) can only access data in slot1. A request for download only returns the data from slot1. A download from the bootloader area and/or slot0 should in all cases be prohibited as these area's contain decrypted images.

PhilippFinke, 2019-06-03:
Thanks, that already helped me and sounds reasonable. Can anybody confirm this?

rakons, 2019-06-03:
Guys. Anyway if we can upload firmware to the slot1 without even pairing it is a big issue in security. Personally, I would not be satisfied if the only protection is that firmware would not be switched.
It makes it easy to destroy the function of the device - you may reset the chip without any authentication. You may kill the battery sending the (broken) firmware over and over again. You may finally destroy the flash slot1 if the battery would last long enough.

PhilippFinke, 2019-06-03:
@rakons, you're right here. That also applies to reading files, stats, ...

PhilippFinke, 2019-06-03:
I had a closer look at the image encryption feature of mcuboot (CONFIG_ BOOT_ENCRYPT_RSA, I think). In my understanding the FW image will be transferred unencrypted and will be encrypted by mcuboot on the fly while writing the image to slot 1. But this keeps the system vulnerable for MITM attacks (a potential attacker can sniff the unencrypted image data while it is uploaded via an unencrypted BT connection).

Laczen, 2019-06-03:
@rakons, we where discussing security. You are talking about something else. Regarding security you need to get the basics right. This is that mcumgr should not be allowed to do anything outside slot1. I do agree extra protection can be put in place but how to do it is something else, I wouldn't be satisfied with a fixed pairing key as this is easy to break.
@PhilippFinke It is not vulnerable to mitm attacks, the firmware is encrypted before sending, it is copied encrypted to slot 1 and mcuboot decrypts when copying to slot 0. When moving to slot 1 (the image that was in slot 0) it is encrypted again. No unencrypted data is ever visible.

rakons, 2019-06-03:
As far as I am following the topic the BT level security here is discussed. So from the BT transport level, it is totally not safe. It does not matter there if the image is encrypted or not. The fact you may destroy the device or you may disallow it to work by sending any broken data is serious risk - I would not like to see for example a drone with bootloader written like that over my head.

Laczen, 2019-06-03:
@rakons, again, get the basics right. What mcuboot does is boot the image in slot 0, you can break the upgrade capabilities of the device by writing to slot 1 but it will not boot the image you have written or will the device be broken. I would be a lot more worried about a bt device that does a secure communication and directly changes the runable image. Would it be better to use a secure communication channel, of course, would it help in providing protection against IP theft, don't count on it.
@PhilippFinke The main idea is: protect your ip by encrypting the image and then distribute it using a open communication or a protected channel. The last step to the device itself can be either protected or not, it is only the last step and you cannot count on this channel being your protection.

carlescufi, 2019-06-03:
@Laczen 

>@PhilippFinke It is not vulnerable to mitm attacks, the firmware is encrypted before sending, it is copied encrypted to slot 1 and mcuboot decrypts when copying to slot 0

The firmware is not encrypted by default when using mcumgr or mcuboot?

Regardless, here I agree with @rakons that no matter what layer of protection MCUboot and MCUmgr provide, there is nothing wrong with being able to enable BLE pairing in the service, especially if you have a hardcoded passkey like @PhilippFinke does.

PhilippFinke, 2019-06-03:
@carlescufi , don't get me wrong here. The passkey on my devices is different on all device instances. I need a way to pass that key when using mcumgr.

rakons, 2019-06-03:
I know what mcuboot is doing. But since the first post, we are talking about mcumgr service. And if it allows anybody to write slot1, it would lead to total slot1 wear if somebody wishes to do it. And you still may reset the device - that may be a huge risk. From that point of view, I am still not feeling save by the fact that you may not switch the slots without proper codes.

It is like leaving doors widely open at night and saying that it is safe as you have all the money in the bank. And I feel like you are me educating again and again how a bank account works.

carlescufi, 2019-06-03:
so, just to be clear: not having pairing required on the BLE service *is* an issue because as @rakons describes, it leads to denial-of-service: you can wear out the flash, drain its battery or keep its BLE connection occupied for long amounts of time, effectively denying the device its right to operate.

joerchan, 2019-06-03:
> I copied the content of smp_bt.h and smp_bt.c and changed the permissions of the single characteristic in the SMP service from BT_GATT_PERM_WRITE to BT_GATT_PERM_WRITE_AUTHEN.

AUTHEN should be set if you need both Encryption and MITM protection.
If you only need encryption then BT_GATT_PERM_WRITE_ENCRYPT should be used.

> When I run mcumgr now, I always get an Insufficient authetication error, which makes sense. I'm running simple-agent on my Linux machine as global agent, that asks me to enter the passkey when I want to connect and pair to my peripheral. Unfortunately this is not the case when using mcumgr.

What exactly do you mean here?
You are not able successfully pair? 
Or are you seeing a bug where you are able to perform the write operation unencrypted even though authentication was required on the attribute?

PhilippFinke, 2019-06-03:
Hey @joerchan,

I want both encryption and MITM protection. So AUTHEN is my right choice here.

I had a look at the btmon output, while using mcumgr. It seems that mcumgr registers its own agent under Linux with the incorrect IO capas for my purpose (plus overriding my own agent). Because of this, I'm not able to enter the correct passkey to pair. This prevents mcumgr to write to the characteristic whose permissions I changed to AUTHEN. This is no bug in my opinion, but prevents the user to force mcumgr to use an authenticated connection.

carlescufi, 2019-06-03:
@PhilippFinke you can try using nRF Connect for iOS or Android instead of `mcumgr` on the command-line. They also support firmware uploads to mcumgr-enabled devices and will definitely set the IO caps correctly

PhilippFinke, 2019-06-03:
@carlescufi I know that this is possible. But this is no option for the final product.

Laczen, 2019-06-03:
@carlescufi, @rakons, I agree that there is nothing wrong with having pairing enabled. I however do not:
a. Agree that this improves IP protection,
b. Think that this is sufficient for a denial of service protection,
and if mcumgr (I don't know if it does) leaves a open door to download data from whatever flash location on the device, then this door should be shut first.

joerchan, 2019-06-03:
@PhilippFinke Looks like the IO cap is the problem then. How to fix that for mcumgr I don't know, I think you will have to take that up with them.

Using a fixed passkey would give you weak authentication, and if you end up using legacy pairing you would have weak encryption as well. Since the passkey would be your encryption key.
Any reason why it needs to be a fixed passkey?
You could use BT_SMP_SC_PAIR_ONLY to disable legacy pairing, or BT_SMP_SC_ONLY to ensure the strongest possible pairing.

PhilippFinke, 2019-06-03:
@joerchan 

I already activated BT_SMP_SC_PAIR_ONLY and BT_SMP_SC_ONLY. I'm using a fixed passkey, because I have no input or output capabilities on my peripheral. A fixed passkey adds a virtual display to my peripheral.

joerchan, 2019-06-03:
@PhilippFinke fair enough. You could also consider OOB authentication, support was recently added.

There is also the possibility to authorize the write request. In the write callback `smp_bt_chr_write` you can return `BT_GATT_ERR(BT_ATT_ERR_AUTHORIZATION)` if you don't want to allow the write.

PhilippFinke, 2019-06-03:
@joerchan 

OOB authentication is also no option. The product has clear specifications and OOB is not part of it. Also when I set the permissions of the SMP BT characteristic to AUTHEN, the behaviour you mentioned happens automatically.

Laczen, 2019-06-03:
@PhilippFinke, I have been studying the smp_svr and mcumgr code in zephyr and it seems that there is no possibility at all to download firmware or bootloader from the device. Your FW and bootloader should be safe.

Laczen, 2019-06-04:
@PhilippFinke, as it seems that mcumgr would require rework. Could you not use the pairing method only to enable the smp_svr which can then be connected to using mcumgr. And to automatically disable after some time. This would solve the reset / flash wear /... issues.
Another possibility is to enable smp_svr only some time after reset.

PhilippFinke, 2019-06-04:
@Laczen, that are good news. Thanks for investigating. I think I will give it a try to register the smp_svr after a paired peer connected to my peripheral and to unregister it after disconnect.

PhilippFinke, 2019-06-04:
I'm struggling a bit with the encryption feature of mcuboot. I see that it can be activated CONFIG_BOOT_ENCRYPT_RSA, but it seems that mcuboot uses a fixed encryption key. Can I change that somehow? And how can I encrypt my FW image with that key?

Laczen, 2019-06-04:
@PhilippFinke, mcuboot uses a fixed encryption key to "encrypt" a randomly generated image encryption key this encrypted random key is added to the image as a TLV. I have never tried to change the fixed encryption key. I would say raise an issue/question in mcuboot.

PhilippFinke, 2019-06-04:
If this is true, the encryption feature doesn't add any security and is completely useless, because an attacker can easily read the fixed key. I raised an issue in mcuboot.

https://github.com/JuulLabs-OSS/mcuboot/issues/489

PR #20557 (2019-12-21): mcumgr: BT SMP transport config for authenticated requirement
This PR introduces a new Kconfig symbol MCUMGR_SMP_BT_AUTHEN.
When selected it configures the Bluetooth mcumgr transport to require
an authenticated connection.

If the Bluetooth mcumgr transport is selected then this new symbol is
selected by default.  Bluetooth SMP is also selected to ensure Zephyr
is configured with Bluetooth security features enabled to provide
Bluetooth authentication APIs to the user's app.  Users can choose to
disable this level of security for the Bluetooth mcumgr transport if
they do not require it.

This P [... cut]
commit b5f335b1 (2019-12-21): mcumgr: BT SMP transport config for authenticated requirement

This commit introduces a new Kconfig symbol MCUMGR_SMP_BT_AUTHEN.
When selected it configures the Bluetooth mcumgr transport to require
an authenticated connection.

If the Bluetooth mcumgr transport is selected then this new symbol is
selected by default.  Bluetooth SMP is also selected to ensure Zephyr
is configured with Bluetooth security features enabled to provide
Bluetooth authentication APIs to the user's app.  Users can choose to
disable this level of security for the Bluetooth mcumgr transport if
they do not require it.

F [... cut]
closed 2019-12-21 by nashif