zephyrproject-rtos/zephyr #16487: tests/kernel/timer/timer_api/kernel.timer  constantly fails in CI on qemu-xtensa

nashif, 2019-05-29:
```
***** Booting Zephyr OS zephyr-v1.14.0-1113-g8806967aed2e *****
Running test suite timer_api
===================================================================
starting test - test_timer_duration_period
PASS - test_timer_duration_period
===================================================================
starting test - test_timer_period_0

    Assertion failed at ./tests/kernel/timer/timer_api/src/main.c:167: test_timer_period_0: tdata.expire_cnt == 1 is false

FAIL - test_timer_period_0
===================================================================
starting test - test_timer_expirefn_null
PASS - test_timer_expirefn_null
===================================================================
starting test - test_timer_periodicity
PASS - test_timer_periodicity
===================================================================
starting test - test_timer_status_get
PASS - test_timer_status_get
===================================================================
starting test - test_timer_status_get_anytime
PASS - test_timer_status_get_anytime
===================================================================
starting test - test_timer_status_sync
PASS - test_timer_status_sync
===================================================================
starting test - test_timer_k_define

    Assertion failed at ./tests/kernel/timer/timer_api/src/main.c:394: test_timer_k_define: tdata.expire_cnt == 4 is false

FAIL - test_timer_k_define
===================================================================
starting test - test_timer_user_data
PASS - test_timer_user_data
===================================================================
starting test - test_timer_remaining_get
PASS - test_timer_remaining_get
===================================================================
Test suite timer_api failed.
===================================================================
PROJECT EXECUTION FAILED
```

andrewboie, 2019-08-13:
I don't think this should be high, only seen with QEMU which is known to have timing issues since we don't use icount

andyross, 2019-08-13:
I'll take spurious bugs, regardless.  Drop to medium for now, will do another pass with a modified rig to rule out timing on tests like this again.