zephyrproject-rtos/zephyr #16492: [Coverity CID :198644]Incorrect expression in /tests/drivers/uart/uart_async_api/src/test_uart_async.c

aasthagr, 2019-05-29:
Static code scan issues seen in File: /tests/drivers/uart/uart_async_api/src/test_uart_async.c
 Category: Incorrect expression
 Function: test_write_abort
 Component: Tests
 CID: 198644
 Please fix or provide comments to square it off in coverity in the link: https://scan9.coverity.com/reports.htm#v32951/p12996

aasthagr, 2019-05-29:
*** CID 198644:  Incorrect expression  (ASSERT_SIDE_EFFECT)
/tests/drivers/uart/uart_async_api/src/test_uart_async.c: 295 in test_write_abort()
289     
290     	uart_rx_enable(uart_dev, rx_buf, sizeof(rx_buf), 50);
291     
292     	uart_tx(uart_dev, tx_buf, 5, 100);
293     	zassert_equal(k_sem_take(&tx_done, 100), 0, "TX_DONE timeout");
294     	zassert_equal(k_sem_take(&rx_rdy, 100), 0, "RX_RDY timeout");
>>>     CID 198644:  Incorrect expression  (ASSERT_SIDE_EFFECT)
>>>     Argument "received" of z_zassert() has a side effect because the variable is volatile.  The containing function might work differently in a non-debug build.
295     	zassert_equal(received, 5, "Incorrect number of bytes received.");
296     	zassert_equal(memcmp(tx_buf, rx_buf, 5), 0, "Buffers not equal");
297     
298     	uart_tx(uart_dev, tx_buf, 95, 100);
299     	uart_tx_abort(uart_dev);
300     	zassert_equal(k_sem_take(&tx_aborted, 100), 0, "TX_ABORTED timeout");

ioannisg, 2019-06-30:
@Mierunski any update on this bug report?

ioannisg, 2019-07-01:
@anangl pls look at this one too, if you have time

ioannisg, 2019-07-10:
Had a look at this one: it's not really a Medium impact bug; it only affects test code, and in fact, it could be solved by finding a way to make the test work with `size_t received` not being declared as volatile.

Reducing to Low; but this needs a fix @Mierunski @anangl  (maybe reveals driver issues)

Mierunski, 2019-07-10:
I will fix it in following days

Mierunski, 2019-07-22:
From what I see, changes that introduced this bug have been removed here: https://github.com/zephyrproject-rtos/zephyr/commit/02d6ea0feb9493165fc12acd8d91543244de0b2f
Bug is no longer present

closed 2019-07-22 by Mierunski