zephyrproject-rtos/zephyr #16506: tests/posix/fs missing ff.h

hakehuang, 2019-05-30:
**Describe the bug**
ff.h file missing

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=board\_xyz
3. make
4. See error

**Expected behavior**
can build pass

**Impact**
fat fs supporting

**Screenshots or console output**
```
/home/shared/disk/zephyr_project/zephyr/tests/posix/fs/src/test_fat_mount.c:8:10: fatal error: ff.h: No such file or directory
 #include <ff.h>
          ^~~~~~
compilation terminated.
CMakeFiles/app.dir/build.make:75: recipe for target 'CMakeFiles/app.dir/src/test_fat_mount.c.obj' failed
make[2]: *** [CMakeFiles/app.dir/src/test_fat_mount.c.obj] Error 1
CMakeFiles/Makefile2:104: recipe for target 'CMakeFiles/app.dir/all' failed
make[1]: *** [CMakeFiles/app.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

```
**Environment (please complete the following information):**
 - OS: (e.g. Linux)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA e9a4ef2b0ae32bb5361a0f802480848520b11cd2

nashif, 2019-06-01:
did you run 'west update'?

hakehuang, 2019-06-03:
update west can fix this issue.

closed 2019-06-03 by hakehuang