zephyrproject-rtos/zephyr #16518: USB UART not working on nrf52840_pca10059

iod, 2019-05-30:
USB UART no longer working on nrf52840_pca10059 after commit 3880a42

carlescufi, 2019-05-30:
@iod can you please post which sample you are using, and how you are flashing the sample?

iod, 2019-05-30:
src/main.c
```c
#include <stdio.h>

void main(void)
{
  printf("hello usb\n");
}
```

prj.conf
```sh
CONFIG_UART_CONSOLE=y
CONFIG_UART_CONSOLE_ON_DEV_NAME="CDC_ACM_0"
CONFIG_UART_LINE_CTRL=y
CONFIG_USB=y
CONFIG_USB_UART_CONSOLE=y
CONFIG_USB_UART_DTR_WAIT=y
```

flashed: with nrfutil dfu usb-serial to nrf52840_pca10059

expected: to be able to connect to device which prints hello usb

actual: ttyACM device no longer shows up after flash

finikorg, 2019-05-31:
I tested `samples/subsys/usb/console` with `reel_board`, works as before.

carlescufi, 2019-05-31:
@iod for completeness, can you please test `samples/subsys/usb/console` on this board to see if this is an issue there?

lemrey, 2019-05-31:
Have tested `samples/subsys/usb/console` on `nrf52840_pca10059`, without the nRF5 booatloder.
The device shows up and I get the expected output on the console.

carlescufi, 2019-05-31:
I tested `samples/subsys/usb/console` on `nrf52840_pca10059` with the nRF5 bootloader flashing with `nrfutil` and it works fine

iod, 2019-05-31:
I have narrowed it down to CONFIG_USB_UART_DTR_WAIT=y that seems to be not working.
@findlayfeng maybe you can add that to the sample to retest?

carlescufi, 2019-05-31:
> I have narrowed it down to CONFIG_USB_UART_DTR_WAIT=y that seems to be not working.
> @finikorg @lemrey @carlescufi maybe you can add that to the sample to retest?

@finikorg any idea why this particular configuration would have stopped working?

anangl, 2019-06-03:
I can only confirm that the problem seems to be the option `CONFIG_USB_UART_DTR_WAIT=y`. It is not specific to nrf52840_pca10059 or to the application. The same behavior (the CDC ACM device does not show up) can be observed also on nrf52840_pca10056, and also with `samples/subsys/usb/console` after adding the mentioned option.

jfischer-phytec-iot, 2019-06-03:
The problem is the init prio (level is APPLICATION anywhere):

CDC ACM class: KERNEL_INIT_PRIORITY_DEVICE 50
uart_console: UART_CONSOLE_INIT_PRIORITY 60
usb_device: APPLICATION_INIT_PRIORITY 90

` uart_console_init` blocks before usb device stack was initialized.

Workaround: set UART_CONSOLE_INIT_PRIORITY to 91

iod, 2019-06-03:
@jfischer-phytec-iot I wanted to confirm that the workaround:
```sh
CONFIG_UART_CONSOLE_INIT_PRIORITY=91
```
fixes the issue the issue for me, thank you.

carlescufi, 2019-06-03:
@jfischer-phytec-iot should we add that line to the sample?

finikorg, 2019-06-04:
I am thinking about following:
```
diff --git a/drivers/console/Kconfig b/drivers/console/Kconfig
index fc89bb83c0..3914f572cb 100644
--- a/drivers/console/Kconfig
+++ b/drivers/console/Kconfig
@@ -53,6 +53,7 @@ config UART_CONSOLE
 
 config UART_CONSOLE_INIT_PRIORITY
        int "Init priority"
+       default 95 if USB_UART_DTR_WAIT
        default 60
        depends on UART_CONSOLE
        help
diff --git a/samples/subsys/usb/console/prj.conf b/samples/subsys/usb/console/prj.conf
index e51d82323b..0961ff4487 100644
--- a/samples/subsys/usb/console/prj.conf
+++ b/samples/subsys/usb/console/prj.conf
@@ -8,3 +8,5 @@ CONFIG_USB_UART_CONSOLE=y
 CONFIG_UART_INTERRUPT_DRIVEN=y
 CONFIG_UART_LINE_CTRL=y
 CONFIG_UART_CONSOLE_ON_DEV_NAME="CDC_ACM_0"
+
+CONFIG_USB_UART_DTR_WAIT=y
```

That default value is only assigned if we put `CONFIG_USB_UART_DTR_WAIT=y` tp `prj.conf`

PR #16613 (2019-06-05): USB: Fix console priority
Fixes #16518
commit df2bb464 (2019-06-05): usb: console: Initialize USB console after USB Device stack

It does make sense to initialize USB console after USB Device stack.
Note that the value is selected only if we specify USB_UART_CONSOLE
in prj.conf, not in menuconfig afterwards.

Fixes #16518
closed 2019-06-05 by carlescufi