zephyrproject-rtos/zephyr #16519: USAGE FAULT occurs when i2c_write is called

icephyr, 2019-05-31:
Hi guys, I am trying to add a sensor driver into zephyr project. The sensor is a pressure sensor named ICP10100 from Invensense, TDK.  The sensor is connected with nRF52832 with I2C.
But USAGE FAULT occurs when I tried to write data to the sensor with the API i2c_write. Here is the detail:
1)During the init flow, I'd like to reset the sensor first. According to the datasheet of ICP10010, I need to send data 0x805d to the sensor by I2C
[image]
so I using the i2c_write API to write data to the sensor, compile it and run, then the usage fault occurs :

> ***** USAGE FAULT *****
  Illegal use of the EPSR
***** Hardware exception *****
Current thread ID = 0x20000118
Faulting instruction address = 0x0
Fatal fault in essential thread! Spinning...

I wonder whether the API is called correctly here and all I want to do is write raw data to the I2C bus.

icephyr, 2019-05-31:
Here is the driver init flow codes， maybe it can help to analyze this problem.

static int icp10100_chip_init(struct device *dev)
{
	unsigned char data_write[10];
	unsigned char data_read[10] = {0};
	int status = 0;
	int out;
	data_write[0] = 0x80;
	data_write[1] = 0x5D;
	status = i2c_write(dev, data_write, 2, DT_INV_ICP10100_0_BASE_ADDRESS);
	if (status)
	{
		printk("i2c_write failed \n");
		return status;
	}
	return 0;
}

int icp10100_init(struct device *dev)
{
	struct icp10100_data *data = dev->driver_data;
	printk("icp10100_init ... \n");
	printk("device_get_binding i2c %s ... \n",DT_INV_ICP10100_0_BUS_NAME);
        data->i2c_master = device_get_binding(DT_INV_ICP10100_0_BUS_NAME);
       if (!data->i2c_master) {
            printk("i2c master not found: %s \n",DT_INV_ICP10100_0_BUS_NAME);
            return -EINVAL;
        }

data->i2c_slave_addr = DT_INV_ICP10100_0_BASE_ADDRESS;

if (icp10100_chip_init(dev) < 0) {
                 return -EINVAL;
           }
return 0;
}

static struct icp10100_data icp10100_data;

DEVICE_AND_API_INIT(icp10100, DT_INV_ICP10100_0_LABEL, icp10100_init, &icp10100_data,
		    NULL, POST_KERNEL, CONFIG_SENSOR_INIT_PRIORITY,
		    &icp10100_api_funcs);

carlescufi, 2019-05-31:
@anangl can you see if you spot anything obvious?

anangl, 2019-05-31:
> @anangl can you see if you spot anything obvious?

Yes.

@icephyr 
The `dev` pointer passed to `icp10100_init` is the pointer to the `struct device` you created for the sensor (the one initialized with `DEVICE_AND_API_INIT(icp10100, ...`). You pass it to `icp10100_chip_init` and then you use it in the call to `i2c_write`. It's wrong, it's a different kind of device. You need to use an I2C device in this call.
See for instance how the tmp007 sensor [is initialized](https://github.com/zephyrproject-rtos/zephyr/blob/master/drivers/sensor/tmp007/tmp007.c#L121).

ioannisg, 2019-05-31:
@icephyr can you check whether the reply covers your concerns/questions? (if so, you could close this issue)

icephyr, 2019-06-01:
thank you guys， I will check it tomorrow and confirm

icephyr, 2019-06-02:
Hi guys, 
    yes, @anangl is right. I passed the wrong parameter to I2C_write. I mistook the API definition, because they have the same structure definition. Now it works after the parameter is revised.

Thanks for your help @carlescufi @anangl

closed 2019-06-02 by icephyr