zephyrproject-rtos/zephyr #16524: tests/drivers/i2c/i2c_api

hakehuang, 2019-05-31:
**Describe the bug**
enable the i2c_api test with attached patch, I find that the twrke18f board missing defines the int2 gpio for the FXOS8700_0

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=twr_ke18f ..
3. make
4. See error

**Expected behavior**
missing in2_gpio define on twrke18f board

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Screenshots or console output**
```
/home/shared/disk/zephyr_project/zephyr/drivers/sensor/fxos8700/fxos8700.c:569:15: error: 'DT_NXP_FXOS8700_0_INT2_GPIOS_CONTROLLER' undeclared here (not in a function); did you mean 'DT_NXP_FXOS8700_0_RESET_GPIOS_CONTROLLER'?
  .gpio_name = DT_NXP_FXOS8700_0_INT2_GPIOS_CONTROLLER,
               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
               DT_NXP_FXOS8700_0_RESET_GPIOS_CONTROLLER
/home/shared/disk/zephyr_project/zephyr/drivers/sensor/fxos8700/fxos8700.c:570:14: error: 'DT_NXP_FXOS8700_0_INT2_GPIOS_PIN' undeclared here (not in a function); did you mean 'DT_NXP_FXOS8700_0_RESET_GPIOS_PIN'?
  .gpio_pin = DT_NXP_FXOS8700_0_INT2_GPIOS_PIN,
              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
              DT_NXP_FXOS8700_0_RESET_GPIOS_PIN
zephyr/drivers/sensor/fxos8700/CMakeFiles/drivers__sensor__fxos8700.dir/build.make:62: recipe for target 'zephyr/drivers/sensor/fxos8700/CMakeFiles/drivers__sensor__fxos8700.dir/fxos8700.c.obj' failed
make[2]: *** [zephyr/drivers/sensor/fxos8700/CMakeFiles/drivers__sensor__fxos8700.dir/fxos8700.c.obj] Error 1
CMakeFiles/Makefile2:2310: recipe for target 'zephyr/drivers/sensor/fxos8700/CMakeFiles/drivers__sensor__fxos8700.dir/all' failed
make[1]: *** [zephyr/drivers/sensor/fxos8700/CMakeFiles/drivers__sensor__fxos8700.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2
```
**Environment (please complete the following information):**
 - OS: (e.g. Linux)
 - Toolchain (e.g Zephyr SDK)
 - Commit SHA

hakehuang, 2019-05-31:
@henrikbrixandersen  @MaureenHelm

hakehuang, 2019-05-31:
[0001-enalbe-i2c_api-for-twrke18f.zip](https://github.com/zephyrproject-rtos/zephyr/files/3240574/0001-enalbe-i2c_api-for-twrke18f.zip)

hakehuang, 2019-05-31:
need config CONFIG_FXOS8700_TRIGGER_NONE=y

hakehuang, 2019-05-31:
[0001-enalbe-i2c_api-for-twrke18f.zip](https://github.com/zephyrproject-rtos/zephyr/files/3240616/0001-enalbe-i2c_api-for-twrke18f.zip)

hakehuang, 2019-05-31:
seems the device driver changed, I need update my test settings

closed 2019-05-31 by hakehuang

henrikbrixandersen, 2019-05-31:
The first part of handling this was already submitted in #16116.