zephyrproject-rtos/zephyr #16529: LTS 1.14.0: sanitycheck: Cannot identify OOT boards and shields

gon1332, 2019-05-31:
**Describe the bug**
_sanitycheck_ cannot find OOT boards and OOT shields, even if `--board_root` is explicitly defined.
Everything is tested on 1.14.0 with https://github.com/zephyrproject-rtos/zephyr/commit/d4163bc17ec4c2779b2e53a6d14cca3986d65d1d cherry-picked from v1.14-branch.

The project is a west-T2 star topology.
The main project is the manifest and has zephyr and other projects as 3rd-parties. 

**To Reproduce**
At first I tried:
```
BOARD_ROOT="${MY_PRJ_ROOT}/zephyr/boards"
"${ZEPHYR_BASE}"/scripts/sanitycheck --board-root "${BOARD_ROOT}" --testcase-root "${MY_PRJ_ROOT}" --tag build --build-only --platform my_oot_board --verbose --inline-logs --outdir /tmp/sanity
```

It failed, reporting that `my_oot_board` is not supported, spitting out the list of zephyr in-tree boards.
I investigated the generated `CMakeCache.txt` file, finding out about the `BOARD_DIR` variable, so my second trial was:
```
BOARD_ROOT="${MY_PRJ_ROOT}/zephyr/boards"
"${ZEPHYR_BASE}"/scripts/sanitycheck --board-root "${BOARD_ROOT}" --testcase-root "${MY_PRJ_ROOT}" --tag build --build-only --platform my_oot_board --verbose --inline-logs --outdir /tmp/sanity -x=BOARD_DIR=${BOARD_ROOT}/arm/my_oot_board
```
This time, the `my_oot_board` is recognized, but the shield needed by the apps cannot be found.

I have inserted some debugging messages in `zephyr/cmake/app/boilerplate.cmake` and I've seen that althought the OOT `boards/shields` directory is listed by `sanitycheck`, the contents are omitted.

**Expected behavior**
Since OOT boards and shields are supported, `sanitycheck` should also be adapted to this kind of structures.

**Impact**
Blockage, since we do not have a working testing environment. Delay, if we choose to move OOT boards and shields in-tree again.

**Environment (please complete the following information):**
 - GNU/Linux
 - Zephyr SDK

**Additional context**
Something else I tried was to move only the OOT shield in-tree, but with no success because of how the corresponding cmake process keeps the shield list.

ioannisg, 2019-05-31:
@SebastianBoe FYI

nwsetec, 2019-06-20:
I've encountered this issue now as well

SebastianBoe, 2019-07-04:
Hi, I believe sanitycheck is not translating it's received
--board-root argument to a CMake argument as it should have been
doing. This should be fixed, but first, as a workaround, could
you try instead of doing,

```
BOARD_ROOT="${MY_PRJ_ROOT}/zephyr/boards"
"${ZEPHYR_BASE}"/scripts/sanitycheck --board-root "${BOARD_ROOT}" --testcase-root "${MY_PRJ_ROOT}" --tag build --build-only --platform my_oot_board --verbose --inline-logs --outdir /tmp/sanity -x=BOARD_DIR=${BOARD_ROOT}/arm/my_oot_board
```

do

`"${ZEPHYR_BASE}"/scripts/sanitycheck --board-root "${BOARD_ROOT}" --testcase-root "${MY_PRJ_ROOT}" --tag build --build-only --platform my_oot_board --verbose --inline-logs --outdir /tmp/sanity -x=BOARD_ROOT=${BOARD_ROOT}
`
I believe this will work because AFAIK BOARD_ROOT is
unfortunately not read as an environment variable. Also,
specifying BOARD_DIR should be unnecessary when BOARD_ROOT is set
appropriately.

gon1332, 2019-07-08:
@SebastianBoe Thanks for the input. Currently I don't have time to test it, but will do as soon as possible.

nashif, 2019-09-19:
BOARD_ROOT needs to point to a directory that has a board directory structure similar to what we have in master, i.e. <oot-root>/boards/<arch>/<board>/...

--board-root of sanitycheck should point to <oot-root/boards>

I think this is where things are failing. Please confirm.

nashif, 2019-11-12:
no response after 2 months

closed 2019-11-12 by nashif