zephyrproject-rtos/zephyr #16545: west: diagnose dependency version failures

pabigot, 2019-06-02:
Putting this here because the cause of breakage was a changed Zephyr file; maybe it belongs in west too/instead.

So I happened to update to current master, and did a `west update` at some point too, and got this with `ninja flash`: 
```
[blah blah blah]
FileNotFoundError: [Errno 2] No such file or directory: 'pyocd': 'pyocd'
```

Appears to be due to #12879 expecting a PyOCD release more recent than 0.12.0, which had been working fine and was what I got when I set up for Zephyr only three months ago.

It'd be more user-friendly if, when Zephyr introduces a dependency on an upstream tool interface change, that it would emit a diagnostic when it finds the previous release, including instructions on how to update.  I only use `pip` because Zephyr requires it indirectly, and IMO should not be expected to know how to do this.

For others who get here and my own notes, it's:
```
pip3 install pyocd -U
```
which will (today) give you pyocd 0.16.1 and also update a bunch of other things for you at no extra charge.

mbolivar, 2019-12-19:
> I only use `pip` because Zephyr requires it indirectly, and IMO should not be expected to know how to do this.

Revisiting this, I think Zephyr requires pip quite directly, and understanding it is a reasonable expectation of our users, at least at this time.

Can we close this?

pabigot, 2019-12-19:
I don't really agree that users should be expected how to use pip, but it looks like there's some stuff in beyond-GSG that adds some clues about `pip -U`.  I suppose that's good enough.

closed 2019-12-19 by pabigot