zephyrproject-rtos/zephyr #16567: [Coverity CID :198880]Possible Control flow issues in /samples/net/lwm2m_client/src/lwm2m-client.c

aasthagr, 2019-06-03:
Static code scan issues seen in File: /samples/net/lwm2m_client/src/lwm2m-client.c
 Category: Possible Control flow issues
 Function: temperature_get_buf
 Component: Samples
 CID: 198880
 Please fix or provide comments to square it off in coverity in the link: https://scan9.coverity.com/reports.htm#v32951/p12996

aasthagr, 2019-06-03:
```
*** CID 198880:  Possible Control flow issues  (DEADCODE)
/samples/net/lwm2m_client/src/lwm2m-client.c: 193 in temperature_get_buf()
187     
188     #if defined(CONFIG_FXOS8700_TEMP)
189     	dev = device_get_binding(DT_NXP_FXOS8700_0_LABEL);
190     #endif
191     
192     	if (dev != NULL) {
>>>     CID 198880:  Possible Control flow issues  (DEADCODE)
>>>     Execution cannot reach this statement: "if (sensor_sample_fetch(dev...".
193     		if (sensor_sample_fetch(dev)) {
194     			LOG_ERR("temperature data update failed");
195     		}
196     
197     		sensor_channel_get(dev, SENSOR_CHAN_DIE_TEMP,
198     				  (struct sensor_value *) &v);
```

mike-scott, 2019-06-05:
Coding style is intentional here.  The plan is to add more sensors as #if checks and end with a check of dev != null.

ioannisg, 2019-07-04:
Then this issue will always trigger the same coverity report, @mike-scott , is there anything we can do? Should we just close it then?

ioannisg, 2019-07-04:
Otherwise, we can keep the issue, but tet's remove the bug, at least, since this is intentional.

carlescufi, 2019-08-28:
Marked as intentional in Coverity

closed 2019-08-28 by carlescufi