zephyrproject-rtos/zephyr #16574: [Coverity CID :198873]Incorrect expression in /tests/drivers/uart/uart_async_api/src/test_uart_async.c

aasthagr, 2019-06-03:
Static code scan issues seen in File: /tests/drivers/uart/uart_async_api/src/test_uart_async.c
 Category: Incorrect expression
 Function: test_chained_read
 Component: Tests
 CID: 198873
 Please fix or provide comments to square it off in coverity in the link: https://scan9.coverity.com/reports.htm#v32951/p12996

aasthagr, 2019-06-03:
```
*** CID 198873:  Incorrect expression  (ASSERT_SIDE_EFFECT)
/tests/drivers/uart/uart_async_api/src/test_uart_async.c: 135 in test_chained_read()
129     				  0,
130     				  "RX_DISABLED occurred");
131     		snprintf(tx_buf, sizeof(tx_buf), "Message %d", i);
132     		uart_tx(uart_dev, tx_buf, sizeof(tx_buf), 100);
133     		zassert_equal(k_sem_take(&tx_done, 100), 0, "TX_DONE timeout");
134     		zassert_equal(k_sem_take(&rx_rdy, 1000), 0, "RX_RDY timeout");
>>>     CID 198873:  Incorrect expression  (ASSERT_SIDE_EFFECT)
>>>     Argument "read_len" of z_zassert() has a side effect because the variable is volatile.  The containing function might work differently in a non-debug build.
135     		zassert_equal(read_len, sizeof(tx_buf),
136     			      "Incorrect read length");
137     		zassert_equal(memcmp(tx_buf, read_ptr, sizeof(tx_buf)),
138     			      0,
139     			      "Buffers not equal");
140     	}
```

ioannisg, 2019-07-17:
@Mierunski could you  please see if this is still a bug, and if so, suggest a fix?

ioannisg, 2019-07-27:
@Mierunski ping

PR #17842 (2019-07-30): tests: drivers: Fix coverity issue 198873
Fix coverity issue 198873

Fixes #16574

I don't know how to check if coverity issues are fixed locally, if there is some test that I should perform please let me know
closed 2019-07-30 by ioannisg