zephyrproject-rtos/zephyr #16587: build failures with gcc 9.x

nashif, 2019-06-03:
**Describe the bug**
warnings with new compilers that turn into errors when building using sanitycheck

```
In file included from /home/nashif/Work/zephyrproject/zephyr/subsys/net/l2/ethernet/arp.c:16:
/home/nashif/Work/zephyrproject/zephyr/include/net/net_pkt.h: In function ‘net_pkt_set_src_ipv6_addr’:
/home/nashif/Work/zephyrproject/zephyr/include/net/net_pkt.h:824:9: error: taking address of packed member of ‘struct net_ipv6_hdr’ may result in an unaligned pointer value [-Werror=address-of-packed-member]
  824 |         &NET_IPV6_HDR(pkt)->src);
      |         ^~~~~~~~~~~~~~~~~~~~~~~
/home/nashif/Work/zephyrproject/zephyr/subsys/net/l2/ethernet/arp.c: In function ‘arp_hdr_check’:
/home/nashif/Work/zephyrproject/zephyr/subsys/net/l2/ethernet/arp.c:537:32: error: taking address of packed member of ‘struct net_arp_hdr’ may result in an unaligned pointer value [-Werror=address-of-packed-member]
  537 |      net_ipv4_is_addr_loopback(&arp_hdr->src_ipaddr)) {
      |                                ^~~~~~~~~~~~~~~~~~~~
/home/nashif/Work/zephyrproject/zephyr/subsys/net/l2/ethernet/arp.c: In function ‘net_arp_input’:
/home/nashif/Work/zephyrproject/zephyr/subsys/net/l2/ethernet/arp.c:590:9: error: taking address of packed member of ‘struct net_arp_hdr’ may result in an unaligned pointer value [-Werror=address-of-packed-member]
  590 |         &arp_hdr->src_ipaddr,
      |         ^~~~~~~~~~~~~~~~~~~~
/home/nashif/Work/zephyrproject/zephyr/subsys/net/l2/ethernet/arp.c:602:30: error: taking address of packed member of ‘struct net_arp_hdr’ may result in an unaligned pointer value [-Werror=address-of-packed-member]
  602 |       net_ipv4_is_addr_mcast(&arp_hdr->src_ipaddr)) {
      |                              ^~~~~~~~~~~~~~~~~~~~
/home/nashif/Work/zephyrproject/zephyr/subsys/net/l2/ethernet/arp.c:608:42: error: taking address of packed member of ‘struct net_arp_hdr’ may result in an unaligned pointer value [-Werror=address-of-packed-member]
  608 |   addr = if_get_addr(net_pkt_iface(pkt), &arp_hdr->dst_ipaddr);
      |                                          ^~~~~~~~~~~~~~~~~~~~
/home/nashif/Work/zephyrproject/zephyr/subsys/net/l2/ethernet/arp.c:635:8: error: taking address of packed member of ‘struct net_arp_hdr’ may result in an unaligned pointer value [-Werror=address-of-packed-member]
  635 |        &arp_hdr->src_ipaddr,
      |        ^~~~~~~~~~~~~~~~~~~~
/home/nashif/Work/zephyrproject/zephyr/subsys/net/l2/ethernet/arp.c:655:27: error: taking address of packed member of ‘struct net_arp_hdr’ may result in an unaligned pointer value [-Werror=address-of-packed-member]
  655 |   if (net_ipv4_is_my_addr(&arp_hdr->dst_ipaddr)) {
      |                           ^~~~~~~~~~~~~~~~~~~~
/home/nashif/Work/zephyrproject/zephyr/subsys/net/l2/ethernet/arp.c:657:8: error: taking address of packed member of ‘struct net_arp_hdr’ may result in an unaligned pointer value [-Werror=address-of-packed-member]
  657 |        &arp_hdr->src_ipaddr,
      |        ^~~~~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
make[3]: *** [zephyr/subsys/net/l2/ethernet/CMakeFiles/subsys__net__l2__ethernet.dir/build.make:63: zephyr/subsys/net/l2/ethernet/CMakeFiles/subsys__net__l2__ethernet.dir/arp.c.obj] Error 1
```
**To Reproduce**

Build for native_posix on Fedora 30 with new compiler

nashif, 2019-06-03:
mostly in networking code...

galak, 2019-08-14:
@nashif Are these fixed now?

jukkar, 2019-08-15:
Regarding my fix proposal for networking at #16587, that PR is working but the changes are not that great and we can do better. @ozhuraki has been investigating this and he will send a new and more sane version.

nashif, 2019-08-21:
partially fix with workaround, lowering priority

stephanosio, 2021-09-03:
This is not resolved, is it?
https://github.com/zephyrproject-rtos/zephyr/blob/e12e126f3262ffcac90572a6be4dffe12f17a30f/cmake/compiler/gcc/compiler_flags.cmake#L45-L50

ozhuraki, 2021-09-03:
@stephanosio

I would say, "_packed" needs to be dropped from the network packet structures (they are already designed so), which would avoid all these (here the compiler is hinting to do so).

cfriedt, 2021-09-18:
@rlubos - can you please chat with @stephanosio about this?

carlescufi, 2021-09-29:
I've investigated this a bit.

It only triggers with `struct in6_addr` and `struct in_addr` because those happen to have alignment requirements (i.e. they are no just structs that hide a simple byte array, but have 16-bit and 32-bit words in them). The possible solutions:

1. Declare `struct in6_addr` and `struct in_addr` as `__packed`. This iis going to bloat the code if those structs are used widely outside the actual places where we parse/prepare a packet to go over the line
2. Change the contents of the IP and ARP `__packed` header structs, so that instead of having a `struct in_addr` they have a `uint8_t src.dst[4]`, and then make sure that we convert to an actual `struct in_addr` whenever required.

cfriedt, 2021-09-29:
> ```
> static inline bool net_ipv6_is_addr_loopback(struct in6_addr *addr)
> {
>         return UNALIGNED_GET(&addr->s6_addr32[0]) == 0 &&
>                 UNALIGNED_GET(&addr->s6_addr32[1]) == 0 &&
>                 UNALIGNED_GET(&addr->s6_addr32[2]) == 0 &&
>                 ntohl(UNALIGNED_GET(&addr->s6_addr32[3])) == 1;
> }
> ```
> 
> This makes no sense, because the param to the function is `struct in6_addr *` which implies an aligned struct, since the struct itself is not declared packed when defined, and we don't take `__packed struct in6_addr *addr` as param to the function. So that means that the `UNALIGNED_GET` in this function is not necessary, at least with the warnings enabled. So the right fix for this function, aside from enabling the warnings, would be to either not use `UNALIGNED_GET` **or** pass in an array of 16 bytes instead of a `struct in6_addr *`.

In Linux, it appears that `struct in6_addr` is not aligned. Alignment would follow the required alignment of the first element in the struct, which is a `u8[]`. The other fields are union with that and do not require additional padding.

https://github.com/torvalds/linux/blob/master/include/uapi/linux/in6.h#L33

The `UNALIGNED_GET()` may not be necessary, I think.

carlescufi, 2021-09-29:
>In Linux (and I think it's actually in general), it appears that struct in6_addr is unaligned.

You probably mean "not packed", because that struct is certainly aligned by the compiler when `__UAPI_DEF_IN6_ADDR_ALT` is enabled.

>Alignment would follow the required alignment of the first element in the struct, which is a u8[]

Not if it's a union that has a member in it with higher alignment requirements, which is the case here.

cfriedt, 2021-09-29:
Ack - i meant alignment actually, but was confused about alignment requirements. You're right, it's not the alignment of the first member that matters.

Note, the Linux definition is *not* packed, but I believe it would require `UNALIGNED_GET()` if it could be in any buffer. Reversing my previous statement.

carlescufi, 2021-09-29:
>Note, the Linux definition is not packed,

I noticed that as well, but I think it's because in Linux they consciously allocate the buffer where the packet will be in an aligned manner:
https://github.com/torvalds/linux/blob/9e9fb7655ed585da8f468e29221f0ba194a5f613/include/linux/skbuff.h#L2894

And then they don't access the header memory directly like we do, but rather take a pointer into the buffer and cast it to a non-packed struct:
https://github.com/torvalds/linux/blob/9e9fb7655ed585da8f468e29221f0ba194a5f613/include/linux/ipv6.h#L96
https://github.com/torvalds/linux/blob/9e9fb7655ed585da8f468e29221f0ba194a5f613/include/linux/skbuff.h#L2547

In Zephyr, however, we make no assumptions (AFAIK) about the alignment of the buffer itself.

cfriedt, 2021-09-30:
True - I would be happy to close the PR I have open. It was really just to investigate the suggestion by @ozhuraki because that was low hanging fruit. 

I found it interesting that no tests actually failed but I wonder if there are additional tests that need to be added.

In any case, what is the resolution here?

carlescufi, 2021-09-30:
>I found it interesting that no tests actually failed but I wonder if there are additional tests that need to be added.

This is likely because packed or no packed all struct members are in fact aligned and the size of the struct matches the on-wire one.

>In any case, what is the resolution here?

Let's discuss this a bit more with @rlubos and @jukkar

rlubos, 2021-09-30:
I'd personally lean towards using RAW buffers within IP/ARP header structs instead of `struct in6_addr/in_addr` and align the usage where needed, but I'm still not sure how much work would this require, I plan to investigate this.

carlescufi, 2021-10-04:
> I'd personally lean towards using RAW buffers within IP/ARP header structs instead of `struct in6_addr/in_addr` and align the usage where needed, but I'm still not sure how much work would this require, I plan to investigate this.

@rlubos 
I actually think we might be able to get away with not packing those but just adding a `BUILD_ASSERT` to verify that they do not deviate from the correct sizes. That said, if we do that, we need to take care about alignment, and make sure that aligned accesses are alway guaranteed, meaning that the packet is always placed at an aligned address.

carlescufi, 2021-12-03:
Reopening since #39192 only fixed the networking side of things

PR #39192 (2021-11-25): net: Remove unpacked structure references from packed structs
This PR proposes a solution for the "unaligned access to the packed struct member warning" issue within networking code.

The generic approach for the problem is similar in most cases covered in this PR:
* Replace `struct in_addr`/`struct in6_addr` within packed structs with raw buffers,
* Provide `_raw` variants of the most commonly used functions (IPv4/6 copy and compare) and use them where applicable,
* ~~For other cases, create a local copy of `struct in_addr`/`struct in6_addr` on stack, copy the IP address from the actual packe [... cut]
PR #40909 (2021-12-10): Fix all instances of -Waddress-of-packed-member
This finally closes #16587 by removing all remaining pieces of code that can trigger the compiler warning after #39192 resolved all the ones in the networking code.

Additional analysis of the problem can be found [here](https://github.com/zephyrproject-rtos/zephyr/issues/16587#issuecomment-930116330).
commit cf111962 (2021-12-10): gcc: Enable address-of-packed-member warnings

After having resolved all of the instances of packed member access,
re-enable the warning.

Fixes #16587.
closed 2021-12-10 by carlescufi