zephyrproject-rtos/zephyr #16594: net: dns: Zephyr is unable to unpack mDNS answers produced by another Zephyr node

lbise, 2019-06-04:
I have 2 Zephyr nodes connected through a SLIP interface. I want one node to resolve the zephyr.local hostname of the other node so I can retrieve it's ip address.
To do this I want to use mDNS so I configured one node as a mDNS resolver and the other as a responder following samples/net/dns_resolve and samples/net/mdns_responder
The request is sent and properly received but the answer cannot be unpacked.
> [net/conn] [DBG] net_conn_input: (0x2000cdb8): Check UDP listener for pkt 0x200139fc src port 46467 dst port 5353 family 10 chksum 0x505b len 38
[net/conn] [DBG] net_conn_input: (0x2000cdb8): [0] match found cb 0x08014bfd ud 0x2000b5dc rank 0x11
[mdns] [DBG] dns_read: (0x2000cdb8): Received 1 query from fe80::200:5eff:fe00:532d
[mdns] [DBG] dns_read: (0x2000cdb8): [1] query AAAA/IN label .zephyr.local (13 bytes)
[mdns] [DBG] dns_read: (0x2000cdb8): mDNS query to our hostname zephyr.local

However the DNS resolve callback returns an error:
> [BRX] [INF] mdns_result_cb: DNS resolve failed
I dug to see exactly what was the problem, first I manually decoded the answer produced by the responder.

Data received:
f2 b6 80 00  00 01 00 01    00 00 00 00  06 7a 65 70
68 79 72 05  6c 6f 63 61    6c 00 00 1c  00 01 06 7a
65 70 68 79  72 05 6c 6f    63 61 6c 00  00 1c 00 01
00 00 00 1e  00 10 fe 80    00 00 00 00  00 00 02 00
5e ff fe 00  53 37

Decoded data:

Header: f2 b6 80 00  00 01 00 01    00 00 00 00
id = f2b6
query params = 8000
Nb question = 1
Nb answer = 1
Nb auth = 0
Nb add = 0

Question:
QNAME
length = 06
label = (zephyr) 7a 65 70 68 79 72
length = 05
label = (local) 6c 6f 63 61 6c
end = 00
QTYPE = 00 1c (AAAA)
QCLASS = 00 01 (IN)

Answer:
length = 06
label = (zephyr) 7a 65 70 68 79 72
length = 05
label = (local) 6c 6f 63 61 6c
end = 00
QTYPE = 00 1c (AAAA)
QCLASS = 00 01 (IN)
TTL = 00 00 00 1e (30)
RDLENGTH = 00 10 (16 bytes)
addr = fe80:0000:0000:0000:0200:5eff:fe00:5337

The problem is that resolve.c has a callback function which ultimately calls subsys/net/lib/dns/dns_pack.c dns_unpack_answer and this function returns -ENOMEM (https://github.com/zephyrproject-rtos/zephyr/blob/2692f2e616c7f0bfcd19fda8010ab0c124f38cf5/subsys/net/lib/dns/dns_pack.c#L85)
This is because the dns_unpack_answer expects the answer to be compressed (as described in RFC https://tools.ietf.org/html/rfc1035#section-4.1.4), if the answer is not compressed it just ignores it and returns -ENOMEM. Basically that means Zephyr is not able to decode another Zephyr's node mDNS answer.
For me the proper way of handling this, would be to check the answer first 2 bits, if b11 then it is compressed, if this is not the case then the answer is a regular label.

lbise, 2019-06-24:
Hello @jukkar. Do you have any input on this problem?

jukkar, 2019-06-24:
I have been busy with other tasks but can look to it next.

jukkar, 2019-06-25:
@lbise Can you test my fix proposal? I only tested with with unit tests so it should work but it would be nice to get ack from you.

lbise, 2019-07-01:
@jukkar Thanks for the fix. I tested it and it works properly

PR #17033 (2019-07-01): Fix dns_answer_unpack() to support non-compressed answers
Fixes #16594
commit ef109375 (2019-07-01): net: dns: Make dns_unpack_answer() to check non-compressed answers

Modify dns_unpack_answer() function to check if the answer is
compressed or not, and return correct values regardless.

Fixes #16594
commit 686f3d21 (2019-07-02): net: dns: Make dns_unpack_answer() to check non-compressed answers

Modify dns_unpack_answer() function to check if the answer is
compressed or not, and return correct values regardless.

Fixes #16594
closed 2019-07-01 by jukkar