zephyrproject-rtos/zephyr #16599: drivers: usb_dc_nrfx: unstable handling of hosts suspend/resume

jfischer-phytec-iot, 2019-06-04:
**Describe the bug**

nRF device (nRF52840) stops to communicate over usb after few suspen/resume cycles.

**To Reproduce**

1. configure `hci_usb` sample as usual
2. optional : enable Logging and set  `USB_DEVICE_LOG_LEVEL` to `USB_DEVICE_LOG_LEVEL_DBG`
3. build and flash `hci_usb` sample
4. connect the board to a Linux host
5. in bluetoothctl, execute `power on`, `scan on`. After a while (about 10 seconds), the device will stop to communicate over usb.

**Screenshots or console output**

```
[00:00:28.366,119] <dbg> usb_device.usb_transfer: transfer start, ep=81, data=0x200051bd, dlen=14
[00:00:28.366,851] <dbg> usb_device.usb_transfer_work: transfer done, ep=81, status=0, size=14
[00:00:31.064,025] <dbg> usb_device.usb_handle_control_transfer: ep 0, status 0
[00:00:31.064,056] <dbg> usb_device.usb_handle_request: ** 0 **
[00:00:31.064,056] <dbg> usb_device.custom_handler: bRequest 0x3, wIndex 0x0
[00:00:31.064,086] <dbg> usb_device.usb_handle_std_device_req: REQ_SET_FEATURE
[00:00:31.068,084] <dbg> usb_bluetooth.bluetooth_status_cb: USB device suspended

```

**If you do not see any `usb_bluetooth.bluetooth_status_cb: USB device suspended`, then the usb host do not rise any suspend/resume requests (e.g. Linux host is behind a VM) and the bug will not appear.**

[image]

```
(gdb) bt
#0  0x00004a36 in usbd_dmareq_process ()
    at /home/fischer/zephyr-project/zephyr/ext/hal/nordic/nrfx/hal/nrf_usbd.h:319
#1  usbd_dmareq_process ()
    at /home/fischer/zephyr-project/zephyr/ext/hal/nordic/nrfx/drivers/src/nrfx_usbd.c:1360
#2  0x0000527a in nrfx_usbd_irq_handler ()
    at /home/fischer/zephyr-project/zephyr/ext/hal/nordic/nrfx/drivers/src/nrfx_usbd.c:1668
#3  0x00007876 in _isr_wrapper () at /home/fischer/zephyr-project/zephyr/arch/arm/core/isr_wrapper.S:120
#4  <signal handler called>
#5  0x3ab551ce in ?? ()
#6  0x4db26158 in ?? ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
```

**Environment (please complete the following information):**
 - OS: Linux 4.19, BlueZ Version 5.50
 - Toolchain  Zephyr SDK
 - Latest master, v1.14

**Workaround:**
Disable Remote Wakeup feature:
```diff
diff --git a/drivers/usb/device/Kconfig b/drivers/usb/device/Kconfig
index 254a255f7b..de6bfa7c71 100644
--- a/drivers/usb/device/Kconfig
+++ b/drivers/usb/device/Kconfig
@@ -64,7 +64,7 @@ menuconfig USB_NRF52840
        select USB_DEVICE_DRIVER
        select HAS_DTS_USB
        select NRFX_USBD
-       select USB_DEVICE_REMOTE_WAKEUP
+       #select USB_DEVICE_REMOTE_WAKEUP
        help
          nRF52840 USB Device Controller Driver
```

carlescufi, 2019-06-04:
I tested this with logging in a VM,  and indeed the device is not being suspended, except at the very beginning but only once.

anangl, 2019-06-19:
@jfischer-phytec-iot Could you try if this patch: https://github.com/anangl/zephyr/commit/0e812743166f4b00977737e6e1388b271f51e41c would solve the problem? It's marked as DNM since such patch should enter through an nrfx update.

Regarding `USB_DEVICE_REMOTE_WAKEUP`, I think that disabling the option could only accidentally prevent the issue as this only makes the device not report to the host that such feature is supported. It does not have an influence on how the device handles suspends and resumes. However, it seems that this option indeed should not be enabled always for nRF devices (via `select`, i.e. without a possibility to disable it), as this will make USB3CV remote wakeup tests impossible to pass. It should be rather enabled when an application actually makes remote wakeup requests, like in the hid_mouse sample for instance.

jfischer-phytec-iot, 2019-06-19:
@anangl The patch does not fix the issue but (as expected) fixes the problem that the driver gets stuck in `usbd_dmareq_process`.

carlescufi, 2019-09-05:
For tracking, a wider description can be found [here](https://github.com/zephyrproject-rtos/zephyr/issues/17488#issuecomment-516800469)

carlescufi, 2019-09-05:
@anangl @emob-nordic Please re-run the testbench and close this issue if appropriate.

emob-nordic, 2019-09-16:
I do not record any abnormalities while testing scenario presented by @jfischer-phytec-iot . The controller acted well, I dont see any bugs here.

closed 2019-09-16 by emob-nordic