zephyrproject-rtos/zephyr #16602: Bluetooth: GATT Discovery: Descriptor Discovery by range Seg Fault

paulgavrikov, 2019-06-04:
**Describe the bug**
Descriptor discovery by range ends in seg fault.

**To Reproduce**
Steps to reproduce the behavior:
1. Modify the discovery code in _central_hr_ sample to 
```		
discover_params.uuid = NULL;
discover_params.func = discover_func;
discover_params.start_handle = 0x0001;
discover_params.end_handle = 0xffff;
discover_params.type = BT_GATT_DISCOVER_DESCRIPTOR;
```

**Expected behavior**
Discovery should work.

**Impact**
Showstopper.

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain: Zephyr SDK 0.10.0
 - Commit SHA or Version used: 2692f2e616c7f0bfcd19fda8010ab0c124f38cf5

Vudentz, 2019-06-04:
Any chance to get a backtrace: addr2line?

aescolar, 2019-06-04:
@paulgavrikov Which platform did you test on?

paulgavrikov, 2019-06-04:
@Vudentz will try. @aescolar nrf52_bsim

aescolar, 2019-06-04:
@paulgavrikov @Vudentz : Then please run that simulated device in valgrind, that will give you a call trace on the segfault.

paulgavrikov, 2019-06-11:
@Vudentz I wasn't able to run it on ble_sim as of now but I tested in on an nRF52840. The discovery callback is triggered one correctly but returning `BT_GATT_ITER_CONTINUE` keeps the callback being called with random handles. It seems like this is an infinite loop.

Example output:

> [14:05:39:103] [ATTRIBUTE] handle 0x0004␍␊ <-- this is ok
[14:05:39:106] [ATTRIBUTE] handle 0x6266␍␊ <-- now we only get random handles
[14:05:39:117] [ATTRIBUTE] handle 0xa7a6␍␊   
[14:05:39:117] [ATTRIBUTE] handle 0x7548␍␊
[14:05:39:117] [ATTRIBUTE] handle 0xb25b␍␊
[14:05:39:117] [ATTRIBUTE] handle 0xb2f3␍␊
[14:05:39:117] [ATTRIBUTE] handle 0xa6fa␍␊
[14:05:39:133] [ATTRIBUTE] handle 0x4bb2␍␊
[14:05:39:133] [ATTRIBUTE] handle 0x8b14␍␊
[14:05:39:133] [ATTRIBUTE] handle 0x43a4␍␊
[14:05:39:133] [ATTRIBUTE] handle 0x5f02␍␊
.....

paulgavrikov, 2019-06-18:
@Vudentz actually, the discovery does finish after a while, in my case 16383 iterations. Interestingly it seems to "discover" attributes multiple times e.g.:

```
[12:21:29:704] [ATTRIBUTE] handle 32767␍␊
[12:21:29:704] [ATTRIBUTE] handle 1152␍␊
[12:21:29:704] [ATTRIBUTE] handle 0␍␊
[12:21:29:704] [ATTRIBUTE] handle 0␍␊
[12:21:29:716] [ATTRIBUTE] handle 0␍␊
[12:21:29:716] [ATTRIBUTE] handle 0␍␊
[12:21:29:716] [ATTRIBUTE] handle 0␍␊
[12:21:29:716] [ATTRIBUTE] handle 0␍␊
[12:21:29:716] [ATTRIBUTE] handle 3␍␊
[12:21:29:716] [ATTRIBUTE] handle 0␍␊
[12:21:29:716] [ATTRIBUTE] handle 0␍␊
[12:21:29:728] [ATTRIBUTE] handle 0␍␊
[12:21:29:728] [ATTRIBUTE] handle 0␍␊
[12:21:29:728] [ATTRIBUTE] handle 0␍␊
[12:21:29:728] [ATTRIBUTE] handle 1␍␊
[12:21:29:728] [ATTRIBUTE] handle 0␍␊
```

Vudentz, 2019-06-18:
Manage to reproduce this one, seems we are getting past the PDU which causes the problem, though I got a strange response from Android:

< ACL Data TX: Handle 3585 flags 0x00 dlen 9                         #1 [hci0] 13:23:17.776067
      ATT: Find Information Request (0x04) len 4
        Handle range: 0x0001-0xffff
> HCI Event: Number of Completed Packets (0x13) plen 5               #2 [hci0] 13:23:17.992344
        Num handles: 1
        Handle: 3585
        Count: 1
> ACL Data RX: Handle 3585 flags 0x02 dlen 22                        #3 [hci0] 13:23:18.040524
      ATT: Find Information Response (0x05) len 17
        Format: UUID-16 (0x01)
        Handle: 0x0001
        UUID: Primary Service (0x2800)
        Handle: 0x0014
        UUID: Primary Service (0x2800)
        Handle: 0x0028
        UUID: Primary Service (0x2800)
        Handle: 0x0033
        UUID: Primary Service (0x2800)

It seems to be responding with only the Primary service which is not what the spec says, or actually don't say so all the attributes types must be returned.

Vudentz, 2019-06-19:
Seems that Android has a bug:

```
< ACL Data TX: Handle 3585 flags 0x00 dlen 9                     #13970 [hci0] 11:17:12.450682
      ATT: Find Information Request (0x04) len 4
        Handle range: 0x0001-0xffff
> HCI Event: Number of Completed Packets (0x13) plen 5           #13971 [hci0] 11:17:12.633040
        Num handles: 1
        Handle: 3585
        Count: 1
> ACL Data RX: Handle 3585 flags 0x02 dlen 9                      #5705 [hci1] 11:17:12.636962
      ATT: Find Information Request (0x04) len 4
        Handle range: 0x0001-0xffff
< ACL Data TX: Handle 3585 flags 0x00 dlen 66                     #5706 [hci1] 11:17:12.654844
      ATT: Find Information Response (0x05) len 61
        Format: UUID-16 (0x01)
        Handle: 0x0001
        UUID: Primary Service (0x2800)
        Handle: 0x0002
        UUID: Characteristic (0x2803)
        Handle: 0x0003
        UUID: Device Name (0x2a00)
        Handle: 0x0004
        UUID: Characteristic (0x2803)
        Handle: 0x0005
        UUID: Appearance (0x2a01)
        Handle: 0x0006
        UUID: Primary Service (0x2800)
        Handle: 0x0007
        UUID: Characteristic (0x2803)
        Handle: 0x0008
        UUID: Service Changed (0x2a05)
        Handle: 0x0009
        UUID: Client Characteristic Configuration (0x2902)
        Handle: 0x000a
        UUID: Characteristic (0x2803)
        Handle: 0x000b
        UUID: Client Supported Features (0x2b29)
        Handle: 0x000c
        UUID: Characteristic (0x2803)
        Handle: 0x000d
        UUID: Database Hash (0x2b2a)
        Handle: 0x000e
        UUID: Characteristic (0x2803)
        Handle: 0x000f
        UUID: EATT Supported (0x2b2b)
> ACL Data RX: Handle 3585 flags 0x02 dlen 27                    #13972 [hci0] 11:17:12.681955
> ACL Data RX: Handle 3585 flags 0x01 dlen 27                    #13973 [hci0] 11:17:12.682374
> ACL Data RX: Handle 3585 flags 0x01 dlen 12                    #13974 [hci0] 11:17:12.682902
      ATT: Find Information Response (0x05) len 61
        Format: UUID-16 (0x01)
        Handle: 0x0001
        UUID: Primary Service (0x2800)
        Handle: 0x0002
        UUID: Characteristic (0x2803)
        Handle: 0x0003
        UUID: Device Name (0x2a00)
        Handle: 0x0004
        UUID: Characteristic (0x2803)
        Handle: 0x0005
        UUID: Appearance (0x2a01)
        Handle: 0x0006
        UUID: Primary Service (0x2800)
        Handle: 0x0007
        UUID: Characteristic (0x2803)
        Handle: 0x0008
        UUID: Service Changed (0x2a05)
        Handle: 0x0009
        UUID: Client Characteristic Configuration (0x2902)
        Handle: 0x000a
        UUID: Characteristic (0x2803)
        Handle: 0x000b
        UUID: Client Supported Features (0x2b29)
        Handle: 0x000c
        UUID: Characteristic (0x2803)
        Handle: 0x000d
        UUID: Database Hash (0x2b2a)
        Handle: 0x000e
        UUID: Characteristic (0x2803)
        Handle: 0x000f
        UUID: EATT Supported (0x2b2b)
```

paulgavrikov, 2019-07-10:
Have to reopen this, works on nrf52840_pca10056 but still issues a seg fault on nrf52_bsim

paulgavrikov, 2019-07-10:
Valgrind:
```
==94554== Thread 2:
==94554== Invalid read of size 1
==94554==    at 0x13D090: bt_uuid_cmp (uuid.c:67)
==94554==    by 0x15798F: bt_gatt_discover (gatt.c:2599)
==94554==    by 0x11F2FD: startGattDiscovery (TDMACCTRL_GATTControl.c:386)
==94554==    by 0x11F140: TDMACCTRL_GATTCmdProcessing (TDMACCTRL_GATTControl.c:287)
==94554==    by 0x11E2DC: ctrlCmdProcessing (TDMACCTRL_CtrlMessageProcessing.c:96)
==94554==    by 0x1232EE: processCommands (RFCP_CtrlInterface.c:193)
==94554==    by 0x122D70: NTCP_processReceivedChannelPacket (NTCP_channelAbstraction.c:185)
==94554==    by 0x122401: slaveThread (CSCTRL_Slave.c:135)
==94554==    by 0x124CA9: z_thread_entry (thread_entry.c:29)
==94554==    by 0x12AD29: posix_thread_starter (posix_core.c:301)
==94554==    by 0x497C3BC: start_thread (pthread_create.c:463)
==94554==    by 0x4A8DE15: clone (clone.S:108)
==94554==  Address 0x0 is not stack'd, malloc'd or (recently) free'd
==94554== 
==94554== 
==94554== Process terminating with default action of signal 11 (SIGSEGV)
==94554==  Access not within mapped region at address 0x0
==94554==    at 0x13D090: bt_uuid_cmp (uuid.c:67)
==94554==    by 0x15798F: bt_gatt_discover (gatt.c:2599)
==94554==    by 0x11F2FD: startGattDiscovery (TDMACCTRL_GATTControl.c:386)
==94554==    by 0x11F140: TDMACCTRL_GATTCmdProcessing (TDMACCTRL_GATTControl.c:287)
==94554==    by 0x11E2DC: ctrlCmdProcessing (TDMACCTRL_CtrlMessageProcessing.c:96)
==94554==    by 0x1232EE: processCommands (RFCP_CtrlInterface.c:193)
==94554==    by 0x122D70: NTCP_processReceivedChannelPacket (NTCP_channelAbstraction.c:185)
==94554==    by 0x122401: slaveThread (CSCTRL_Slave.c:135)
==94554==    by 0x124CA9: z_thread_entry (thread_entry.c:29)
==94554==    by 0x12AD29: posix_thread_starter (posix_core.c:301)
==94554==    by 0x497C3BC: start_thread (pthread_create.c:463)
==94554==    by 0x4A8DE15: clone (clone.S:108)
==94554==  If you believe this happened as a result of a stack
==94554==  overflow in your program's main thread (unlikely but
==94554==  possible), you can try to increase the size of the
==94554==  main thread stack using the --main-stacksize= flag.
==94554==  The main thread stack size used in this run was 8388608.
==94554== 
==94554== HEAP SUMMARY:
==94554==     in use at exit: 2,849 bytes in 15 blocks
==94554==   total heap usage: 101 allocs, 86 frees, 27,238 bytes allocated
==94554== 
==94554== LEAK SUMMARY:
==94554==    definitely lost: 0 bytes in 0 blocks
==94554==    indirectly lost: 0 bytes in 0 blocks
==94554==      possibly lost: 1,224 bytes in 9 blocks
==94554==    still reachable: 1,625 bytes in 6 blocks
==94554==         suppressed: 0 bytes in 0 blocks
==94554== Rerun with --leak-check=full to see details of leaked memory
==94554== 
==94554== For counts of detected and suppressed errors, rerun with: -v
==94554== ERROR SUMMARY: 1 errors from 1 contexts (suppressed: 0 from 0)
Segmentation fault (core dumped)
```

PR #16925 (2019-06-24): Bluetooth: GATT: Fix bt_gatt_discovery with BT_GATT_DISCOVER_DESCRIPTOR
Fix iterating past the response which causes an invalid memory to be
accessed and passed over to the callback as if there were more
attributes found.
    
Fixes #16602

Note: Only the first patch need to be backported, the second was used just for checking if things are working properly.
commit 0d0e9d6b (2019-06-24): Bluetooth: GATT: Fix bt_gatt_discovery with BT_GATT_DISCOVER_DESCRIPTOR

Fix iterating past the response which causes an invalid memory to be
accessed and passed over to the callback as if there were more
attributes found.

Fixes #16602
commit eb629627 (2019-06-24): Bluetooth: GATT: Fix bt_gatt_discovery with BT_GATT_DISCOVER_DESCRIPTOR

Fix iterating past the response which causes an invalid memory to be
accessed and passed over to the callback as if there were more
attributes found.

Fixes #16602
commit eb8abda6 (2019-06-24): Bluetooth: GATT: Fix bt_gatt_discovery with BT_GATT_DISCOVER_DESCRIPTOR

Fix iterating past the response which causes an invalid memory to be
accessed and passed over to the callback as if there were more
attributes found.

Fixes #16602
closed 2019-07-10 by jhedberg