zephyrproject-rtos/zephyr #16603: Bluetooth: Gatt Discovery: BT_GATT_DISCOVER_PRIMARY returns all services while BT_GATT_DISCOVER_SECONDARY returns none

paulgavrikov, 2019-06-04:
**Describe the bug**
Setting the GATT discovery type for a range search (i.e. UUID = NULL) to `BT_GATT_DISCOVER_PRIMARY ` returns all services while `BT_GATT_DISCOVER_SECONDARY ` returns none at all.

**To Reproduce**
Steps to reproduce the behavior:
1. Modify the discovery code in _central_hr_ sample to 
```		
discover_params.uuid = NULL;
discover_params.func = discover_func;
discover_params.start_handle = 0x0001;
discover_params.end_handle = 0xffff;
discover_params.type = BT_GATT_DISCOVER_PRIMARY;
```
or 
```
discover_params.type = BT_GATT_DISCOVER_SECONDARY;
```

**Expected behavior**
Primary discovery should only return primary services, secondary only secondary or get rid of secondary discovery and just have a new type `BT_GATT_DISCOVER_SERVICE`

**Impact**
Annoyance.

**Environment (please complete the following information):**
 - OS: Kubuntu 18.04
 - Toolchain: Zephyr SDK 0.10.0
 - Commit SHA or Version used: 2692f2e616c7f0bfcd19fda8010ab0c124f38cf5

Vudentz, 2019-06-04:
We are setting the UUID of Read By Group Request to either primary or secondary, perhaps the remote database don't have a secondary?

paulgavrikov, 2019-06-04:
@Vudentz yeah ... sorry I mixed things up

closed 2019-06-04 by paulgavrikov