zephyrproject-rtos/zephyr #16604: Zephyr fails to build with CPU load measurement enabled

pdunaj, 2019-06-04:
**Describe the bug**
Cannot build with `TRACING_CPU_STATS` enabled as `smp_init` is both external linkage from kernel and internal in ble subsystem.

**To Reproduce**
Steps to reproduce the behavior:
1. Find some bluetooth enabled sample
2. enable `TRACING_CPU_STATS`
3. build

**Expected behavior**
should just work

**Impact**
annoyance, debug tool is unusable

**Screenshots or console output**
```
/home/pdunaj/work/ncs/zephyr/subsys/bluetooth/host/smp.c:2278:12: error: conflicting types for 'smp_init'
 static int smp_init(struct bt_smp *smp)
            ^~~~~~~~
In file included from /home/pdunaj/work/ncs/zephyr/arch/arm/include/kernel_arch_data.h:34,
                 from /home/pdunaj/work/ncs/zephyr/kernel/include/kernel_structs.h:68,
                 from /home/pdunaj/work/ncs/zephyr/subsys/debug/tracing/include/tracing_cpu_stats.h:10,
                 from /home/pdunaj/work/ncs/zephyr/include/tracing.h:34,
                 from /home/pdunaj/work/ncs/zephyr/include/kernel.h:4890,
                 from /home/pdunaj/work/ncs/zephyr/include/zephyr.h:18,
                 from /home/pdunaj/work/ncs/zephyr/subsys/bluetooth/host/smp.c:13:
/home/pdunaj/work/ncs/zephyr/kernel/include/kernel_internal.h:238:13: note: previous declaration of 'smp_init' was here
 extern void smp_init(void);
             ^~~~~~~~
```

**Environment (please complete the following information):**
master

**Additional context**
N/A

pdunaj, 2019-06-04:
Seems there is a conflict of smp acronym usage :)

carlescufi, 2019-06-04:
We need to figure out who gets to keep the fancy `smp_init()` name

jhedberg, 2019-06-04:
I think the Bluetooth one could be renamed to `smp_conn_init()` or similar

andyross, 2019-06-04:
Actually this looks like the bug is in the kernel headers.  It's a static function, it shouldn't be in a header to begin with.   Git tells me this came in with the original SMP patch where smp_init was already static.  Looks like prehistorical cruft where I needed it somewhere else or something?

PR #16640 (2019-06-05): kernel/smp: Rename smp_init()
This name collides with one in the bt subsystem, and wasn't named in
proper zephyrese anyway.

Fixes #16604
commit a12f2d66 (2019-06-05): kernel/smp: Rename smp_init()

This name collides with one in the bt subsystem, and wasn't named in
proper zephyrese anyway.

Fixes #16604
closed 2019-06-05 by nashif