zephyrproject-rtos/zephyr #16606: Fault in CPU stats

pdunaj, 2019-06-04:
**Describe the bug**
```
ASSERTION FAIL [current_thread == k_current_get()] @ ./subsys/debug/tracing/cpu_stats.c:126
```

**To Reproduce**
Steps to reproduce the behavior:
1. Enable `TRACING_CPU_STATS`
2. Try to run...

**Expected behavior**
Should work.

**Impact**
annoyance - profiling tool is unusable

**Screenshots or console output**
Above.

**Environment (please complete the following information):**
master

**Additional context**
N/A

pdunaj, 2019-06-04:
It seems that `sys_trace_thread_switched_out` happened before `sys_trace_thread_switched_in`...

pdunaj, 2019-06-04:
I added a null-check on first switched out (and ignore it). After that CPU load seems to show the valid data (as far as I can tell).

nashif, 2019-06-04:
This was introduced by @MarekPieta originally, can you please look at this?

MarekPieta, 2019-06-05:
Yes, no problem. I will take look at it.

MarekPieta, 2019-06-06:
I created a PR fixing this

PR #16655 (2019-06-07): debug: tracing: Fix tracing hooks
Change removes tracing hooks before threads are initialized and thread switched out hook for ARM before first time switching to main thread.

Fixes: #16606
closed 2019-06-07 by nashif