zephyrproject-rtos/zephyr #16631: SDK_VERSION

imadeveloper247, 2019-06-05:
building under Ubuntu 18.04

I think that the file ~zephyrproject/zephyr/cmake/toolchain/zephyr/generic.cmake might have an error on line 3:
     if(NOT DEFINED SDK_VERSION)
perhaps that should be:
    if(NOT SDK_VERSION)

Also the build docs for linux should have us:

export SDK_VERSION=0.10.0

Hope this is of some help!

Phil

carlescufi, 2019-06-05:
@imadeveloper247 you don't need to export the SDK version at all. This is checked by the CMake files, you should only need to export:

`ZEPHYR_SDK_INSTALL_DIR=/path/to/sdk`
`ZEPHYR_TOOLCHAIN_VARIANT=zephyr`

What are you trying to solve exactly?

SebastianBoe, 2019-06-12:
Closing due to reporter being unresponsive.

closed 2019-06-12 by SebastianBoe