zephyrproject-rtos/zephyr #16661: Symmetric multiprocessing (SMP) for ARC HS cores

abrodkin, 2019-06-06:
This issue is to track development of SMP for ARC HS cores.

# Some background
## Supported ARC processors

[ARC EM cores](https://www.synopsys.com/designware-ip/processor-solutions/arc-processors/arc-em-family.html) are single-processor designs yet so only [ARC HS family](https://www.synopsys.com/designware-ip/processor-solutions/arc-processors/arc-hs-family.html) will be supported for now.

ARC HS family allows for designs with up-to 4 cores in one SMP cluster. Typical example might be HS38x4 found in [HSDK board](https://www.synopsys.com/dw/ipdir.php?ds=arc-hs-development-kit).

## Hardware features required for efficient SMP support

The following are features of ARC HS cores useful for SMP-aware software:
 * Shared external memory (DDR) or Shared-Cluster-Memory (SCM)
   As compared to xCCMs (Instruction/Data Closely-Coupled Memory) typically used in deeply embedded designs external memory is available to all the cores in cluser, while xCCMs are private per-core memories.
 * Interrupt Distribution Unit (ARConnect IDU) which allows to route interrupts from peripherals to the particular CPU in the cluser.
 * Global Free-Running Counter (ARConnect GFRC) which gives a unified clock source, rather than having per-core counters.
 * Inter-Core Debug Unit (ARConnect ICD) which allows selectively halt, run and reset cores in the cluster. 
 * Inter-Core Interrupt Unit (ARConnect ICI) which might be used for notifications and remote code execution

And so to support efficient SMP on ARC we need to support most of the HW features listed above. Well with exception of shared memory (DDR controller is supposed to be be set by boot-ROM anyways and SCM requires no configuration - that's just on-chip SRAM) and for starters ICI (relying on random IRQs from peripherals similarly to what happens on Xtensa).

## Supported platforms

### Platforms avaialble today

The following platforms have multi-core (SMP) ARC HS-based SoCs and "SoCs":
 1. HSDK as mentioned above. This is our primary development platfrom for ARC HS. It's available from Synopsys (see https://www.synopsys.com/apps/arc-shopping-cart/) and should be in stock ATM.
 2. MetaWare Debugger. Synopsys propriertary debugger capable of multi-core setup simulation.

Ok I know it doesn't sound right and fair to force developers of open source project to spend a lot of money on proprietary stuff so...

### Platfroms we're actively working on
QEMU. It's been in works for quite some time and Uni-Porcessor (UP) support is quite solid. I'm still hoping we get it published very-very soon. And hopefully by the time we send our patches for upstream review we'll get SMP support for ARC there too.

# Changes required in Zephyr

Given hardware discussed above and my study of Zephyr's kernel code that's the list of atomic changes we'll need to do to get this issue closed:

- [ ] ARC: Add support for ARC HS family of CPUs
- [ ] ARC: Add support for ARC HS family on nSIM
- [ ] ARC: Add HSDK board support
- [ ] ARC: Add QEMU board for ARC
- [ ] ARC: Add support for Interrupt Distribution Unit
- [ ] ARC: Add support for Global Free-Running Counter
- [ ] ARC: Add support for Inter-Core Debug Unit
- [ ] ARC: Add support for Inter-Core Interrupt Unit (ICI)
- [ ] ARC: Implement z_arch_curr_cpu()
- [ ] ARC: Implement smp_init()/z_smp_init()
- [ ] ARC: Implement smp_timer_init()
- [ ] ARC: Implement z_arch_switch()
- [ ] ARC: Enable SMP on capable platfroms
 
Any comments are much apreciated.

abrodkin, 2019-06-06:
@nashif may we get this one assigned to me and @vonhust and if possible targeted to the next release (1.15)?

abrodkin, 2019-06-07:
@nashif is the next milestone 2.0 but not 1.15?

vonhust, 2019-08-10:
This issue can be closed as ARC SMP support is merged

closed 2019-08-10 by vonhust

abrodkin, 2019-08-13:
@vonhust I guess we may mark "ARC: Add support for Inter-Core Interrupt Unit (ICI)" as done as well.
We do have implementation of `z_arch_sched_ipi()`/`sched_ipi_handler()` in https://github.com/zephyrproject-rtos/zephyr/blob/master/arch/arc/core/arc_connect.c or something extra is still needed?