zephyrproject-rtos/zephyr #16670: Memory reports do not work when Nordic proprietary LL is selected

pdunaj, 2019-06-07:
**Describe the bug**
```
pdunaj@ubuntu:~/work/ncs/nrf/applications/nrf_desktop$ time ( B=nrf52840_pca20041; cd build-$B && ninja rom_report )
[1/2] cd /home/pdunaj/work/ncs/nrf/applications/nrf_desktop/build-nrf52840_pca20041/zephyr/cmake...hyr-eabi-nm -o /home/pdunaj/work/ncs/nrf/applications/nrf_desktop/build-nrf52840_pca20041/zephy
FAILED: zephyr/cmake/reports/CMakeFiles/rom_report 
cd /home/pdunaj/work/ncs/nrf/applications/nrf_desktop/build-nrf52840_pca20041/zephyr/cmake/reports && /usr/bin/python3 /home/pdunaj/work/ncs/zephyr/scripts/footprint/size_report -F --objdump /opt/zephyr-sdk/arm-zephyr-eabi/bin/arm-zephyr-eabi-objdump --objcopy /opt/zephyr-sdk/arm-zephyr-eabi/bin/arm-zephyr-eabi-objcopy --nm /opt/zephyr-sdk/arm-zephyr-eabi/bin/arm-zephyr-eabi-nm -o /home/pdunaj/work/ncs/nrf/applications/nrf_desktop/build-nrf52840_pca20041/zephyr
Traceback (most recent call last):
  File "/home/pdunaj/work/ncs/zephyr/scripts/footprint/size_report", line 450, in <module>
    main()
  File "/home/pdunaj/work/ncs/zephyr/scripts/footprint/size_report", line 442, in main
    base + '/', None)
  File "/home/pdunaj/work/ncs/zephyr/scripts/footprint/size_report", line 181, in generate_target_memory_section
    symbols_paths = dict(load_symbols_and_paths(bin_nm, elf_file_abs, source_dir))
  File "/home/pdunaj/work/ncs/zephyr/scripts/footprint/size_report", line 35, in load_symbols_and_paths
    symbol, path = parse_symbol_path_pair(line)
  File "/home/pdunaj/work/ncs/zephyr/scripts/footprint/size_report", line 80, in parse_symbol_path_pair
    assert len(fields) >= 4
AssertionError
ninja: build stopped: subcommand failed.
```

**To Reproduce**
Steps to reproduce the behavior:
1. enable `BT_LL_NRFXLIB`
2. run `ninja rom_report`

**Expected behavior**
Should present a memory report.

**Impact**
annoyance - profiling tool is unusable

**Screenshots or console output**
check above

**Environment (please complete the following information):**
ncs master

**Additional context**
N/A

pdunaj, 2019-06-07:
I am creating a bug upstream as it seems to me the problem may be with how symbols are obtained and not be specific to this library.

ioannisg, 2019-07-03:
@SebastianBoe can you take a look at this bug report from Pawel?

SebastianBoe, 2019-07-03:
Reproduced, will investigate.

ioannisg, 2019-07-04:
Thanks @SebastianBoe

PR #17348 (2019-07-05): size_report: Don't crash when observing stripped symbols
When stripped symbols are present in the build, "size_report" has crashed
due to it not being aware of this case.

This patch causes stripped symbols to be ignored, allowing
analysis of the non-stripped symbols.

It is left as future work to somehow display how much space the stripped
symbols are taking up.

Stripped symbols can be present when third-party binaries are linked
into the build.

This fixes #16670
closed 2019-07-05 by nashif