zephyrproject-rtos/zephyr #16672: nrf: spi: Excess current

chris-makaio, 2019-06-07:
I'm seeing high current when using nrf's SPI driver.
Code running w/o SPI: 220uA (okay given the peripherals)
Code running with SPI: 1.5mA

It can't be my implementation of the sensor's driver because it is still running.
It's a LIS2MDL (custom driver) which works via SPI and I2C - and one driver instance is still running for I2C.

It also seems that there is no DEVICE PM for SPI, is that true?

chris-makaio, 2019-06-07:
So, meanwhile I tried:

- use SPIM2 instead of SPIM3 (because it looks to me that SPIM3 is a bit special with all the "EXTENDED" stuff). 
- try anomalies 198 and 109
- integrated a new "spi_nrfx_sleep" which does spi_nrfx_release and nrfx_spim_unit
- moved "return;"s until I found the line at which the current raises. It's nrf_spim_task_trigger(p_spim, NRF_SPIM_TASK_START);

Nothing helped yet...

chris-makaio, 2019-06-07:
Just noticed I made a mistake in my code and didn't actually switch to SPIM2.
Fixed and now (when using my own sleep function) everything works fine. 

For reference, here's the simple code:
```
static int spi_nrfx_sleep(struct device *dev,
						  const struct spi_config *spi_cfg)
{
	spi_nrfx_release(dev, spi_cfg);
	nrfx_spim_uninit(&get_dev_config(dev)->spim);

	LOG_ERR("Will sleep");
	return 0;
}
```

anangl, 2019-06-07:
#16548 adds power management to SPIM and SPI drivers. @chris-makaio could you try if it works for you?

chris-makaio, 2019-07-18:
@anangl, I was just able to test the I2C parts. It works pretty well.
Here are the results (I know people are looking for possible current consumption figures) for our custom board:
I2C1 active (DEVICE_PM_ACTIVE_STATE): 8uA
I2C1 suspended (DEVICE_PM_SUSPEND_STATE): 3uA 

I was not able to test the SPI yet, though, so if possible, please leave this issue open

chris-makaio, 2019-07-18:
seems #16548 fixed it.
I can't even see a difference between active and suspended (testing with a W25Q spi flash @ SPI2).
No matter what I do, I see 4.2uA

Nice

closed 2019-07-18 by chris-makaio