zephyrproject-rtos/zephyr #16673: usb_dc_stm32: If i remove the cable while writing, the program will freeze.

KwonTae-young, 2019-06-07:
**Describe the bug**
If I remove the cable while writing using `hid_int_ep_write()`, the program will stop.

The `usb_dc_ep_write()` function attempts to get the semaphore using `k_sem_take()`.
`k_sem_take()` always fails if the USB cable is removed in the middle while writing.
I forced `k_sem_give()` to be called if `k_sem_take()` failed a certain number of times.
And I confirmed that the symptoms were resolved.

https://github.com/zephyrproject-rtos/zephyr/blob/master/drivers/usb/device/usb_dc_stm32.c#L738-L742
[image]

**To Reproduce**
```
west init zephyrproject -m https://github.com/KwonTae-young/zephyr/ --mr stm32_usb_sem
cd zephyrproject/zephyr
git checkout 59606eaf94cf91e3f0a346f1e6499584c6b6d257
source zephyr-env.sh
mkdir samples/stm32_usb_sem/build
cd samples/stm32_usb_sem/build
cmake -DBOARD=stm32f4_disco ..
make -j8 flash
```

When connecting USB, "12345678901234567890" is repeatedly written.
If you remove the USB cable in the middle, the program will stop.
Below is a recap video.
https://youtu.be/qIGH5OoDnUU

**Expected behavior**
If I unplug the USB cable suddenly during the write, the program should work normally.

**Impact**
In this situation, the program will stop.

**Screenshots or console output**
1) console
```
***** Booting Zephyr OS zephyr-v1.13.0-7674-g882173ce828a *****
stm32 usb sem example(stm32f4_disco)
100ms repeat print(1)
100ms repeat print(2)
100ms repeat print(3)
100ms repeat print(4)
100ms repeat print(5)
100ms repeat print(6)
100ms repeat print(7)
100ms repeat print(8)
100ms repeat print(9)
100ms repeat print(10)
100ms repeat print(11)
100ms repeat print(12)
100ms repeat print(13)
100ms repeat print(14)
100ms repeat print(15)
100ms repeat print(16)
100ms repeat print(17)
100ms repeat print(18)
100ms repeat print(19)
100ms repeat print(20)
100ms repeat print(21)
100ms repeat print(22)
100ms repeat print(23)
100ms repeat print(24)
100ms repeat print(25)
100ms repeat print(26)
100ms repeat print(27)
100ms repeat print(28)
/********** Unplug the USB cable suddenly **********/
(The program is stopped.)
```

2) windows notepad(HID output)
```
12345678901234567890
12345678901234567890
12345678901234567890
1234567890123
```

**Environment (please complete the following information):**
 - OS: Ubuntu 18.04
 - Toolchain: Zephyr SDK 0.10.0
 - Commit SHA: bd001081fce1e2b2123f8248be9c3c938d1f470c

**Additional context**
As mentioned in `Describe the bug`, I made a temporary commit to fix the symptoms.
The symptom is solved through this commit.(https://github.com/KwonTae-young/zephyr/commit/882173ce828a640a735e629be236f84d8eadfbdd)
```
west init zephyrproject -m https://github.com/KwonTae-young/zephyr/ --mr stm32_usb_sem
cd zephyrproject/zephyr
git checkout 882173ce828a640a735e629be236f84d8eadfbdd
source zephyr-env.sh
mkdir samples/stm32_usb_sem/build
cd samples/stm32_usb_sem/build
cmake -DBOARD=stm32f4_disco ..
make -j8 flash
```

Below is a video with improved symptoms using a temporary commit.
https://youtu.be/CQhCXeNStoE

I'm sorry I do not have enough English. :)
Thank you.

ioannisg, 2019-08-27:
@loicpoulain is this issue being looked at for v2.0.0 release?

loicpoulain, 2019-08-27:
@ioannisg, I'll look at it tomorrow, it's something which should be easily fixable. I need to fix #18717 before.

loicpoulain, 2019-08-28:
@KwonTae-young what is the issue exactly? I'm able to reproduce your behavior, the program is pending in usb_write while USB is disconnected, but once reconnected, it continues...

KwonTae-young, 2019-08-29:
@loicpoulain, I think `usb_write()` should return an error, not a hold.
The USB cable may not be reconnected in some cases.

loicpoulain, 2019-08-29:
@KwonTae-young, agree, the driver usb_ep_write() returns -EAGAIN, but usb_write() loop in that case, it to workaround wrong usage of usb_write which should be only called if the previous one finished (and announced via the usb ep callback), so calling usb_write in an unconditional loop is probably not a good idea. However, it also highlights an issue with usb_dc_stm32 controller driver, since we should be able to detect the disconnect and return a value different from -EAGAIN in case of write while disconnected.

carlescufi, 2019-09-18:
@loicpoulain will you send a PR for this?

erwango, 2019-09-19:
Is that somehow linked to https://github.com/zephyrproject-rtos/zephyr/pull/16026?

erwango, 2019-11-01:
@ydamigos, @loicpoulain any update on this?

erwango, 2019-11-21:
@ydamigos, @loicpoulain any of you working on this one?

loicpoulain, 2019-11-21:
@erwango, let me check that.

loicpoulain, 2019-11-21:
This specific issue (locked in usb_write loop) should is now fixed, thanks to #18646 (77125eafa9e23fbac78809483be9ece1827e36b5)

erwango, 2019-11-21:
> @erwango This specific issue (locked in usb_write loop) should is now fixed, thanks to #18646 ([77125ea](https://github.com/zephyrproject-rtos/zephyr/commit/77125eafa9e23fbac78809483be9ece1827e36b5))

Thanks @loicpoulain !
@KwonTae-young, Can you confirm this is now working on your side?

KwonTae-young, 2019-11-22:
@erwango
I checked on the master branch and it works fine.

erwango, 2019-11-22:
Thanks @KwonTae-young. Closing the point.
Fixed by https://github.com/zephyrproject-rtos/zephyr/pull/18646

closed 2019-11-22 by erwango