zephyrproject-rtos/zephyr #16676: Settings enhancements

Laczen, 2019-06-07:
**Is your enhancement proposal related to a problem? Please describe.**
The settings module is limiting the usability of zephyr in real world applications because of high RAM usage. This high RAM usage is mainly in the modules that are using settings and is a result of the one time loading of all settings at the start.

Apart from this problem there are several improvement opportunities in the settings module

**Describe the solution you'd like**

- [ ] The ability to load part of the settings on demand

- [ ] Processing of the variable names in a less RAM hungry method by keeping the variable name in ROM (flash) as long as possible and not convert it to a list of arguments.

- [ ] The ability to register a handler for a variable directly

- [ ] The possibility to store the complete settings_handler struct in ROM

**Further improvement possibilities**

- [ ] Try to avoid any commit routines by only returning the last valid values

- [ ] Add a delete property that allows to delete all entries in a tree

These future enhancements are probably easiest handled in the backend.

**Proposal**

All the improvements will work around two basic routines that are available to the settings module and to the users of the settings module:
a. ```int settings_name_cmp(const char *name, const char *key, const char **next)``` that compares the start of `name` with `key` and returns the remaining of name (without leading /) in `next`.
b. ```const char *settings_name_skip(const char *name, char *argv)``` that returns the a pointer to the remaining of name (without leading /) and sets argv to a string equal to the start of name terminated by '/0'. This routine should only be used when dealing with dynamic names (e.g. when name starts with a unknown part), otherwise the routine under a. should be used.

optionally b. could be replaced with a classic argc, argv split up, but if possible this should be avoided (because it uses more RAM, and there might be a situation where a dynamic name is followed by a "constant name").

To first improvement will be handled as follows:
a. A settings_handler for a on demand load disables itself from being processed at start (using a flag inside the settings_handler).
b. The settings_load command is changed to take an optional key parameter that filters out any names that do not start with key, if a match is found it calls the settings_handler.
c. The on demand load is done by:
1. enabling the settings_handler
2. calling settings_load with a key that matches what needs to be loaded (e.g. bt/ccc/0xff24...),
the key is generated by a connection request.
3. disables the settings_handler

It is also possible to add a on demand load flag to the settings_handler structure and not to process on demand settings_handlers when calling settings_load at the start. But his would force the user to register a direct handler for on demand loads. In the case it is handled by a flag the on demand loading could also be handled by a "higher level" settings_handler.

The second improvement is done by not splitting up the names.

The third improvement is done by allowing registration of handlers including the separator, the comparison of a handler_name is done using ```settings_name_cmp```.

The fourth improvement could easily be achieved. The separate registration of bt_settings_handlers can be removed by replacing the bt_set with a routine that does a settings_name_cmp with "btmesh", "ccc", ... and calling the already available set routines while "eating" away the name with settings_name_cmp().

Laczen, 2019-06-07:
@rakons, @pdunaj, @nvlsianpu, @carlescufi, @jhedberg, @Vudentz, @MaureenHelm, @fnde-ot

nvlsianpu, 2019-06-07:
@Laczen We are awaiting on your PR-mod to Radek's PR #16609. I understood that form the dev-meeting we have.

Laczen, 2019-06-07:
Of course, this is why I have written down the above. It is what I'm going to do and how it would be used, if anybody would like to comment, say "that is not a good idea", "this does not fit my needs", ... feel welcome.

nvlsianpu, 2019-06-07:
I slight not get how enable/disable settings_handler  will work - bat that's NOP as you want to show this in PR. I see nothing against you want achieve.

Laczen, 2019-06-07:
@nvlsianpu, In the beginning of the settings_handler there will be a check for a flag, if this flag is not set it jumps out of the settings_handler. I guess @rakons has to do something similar to avoid doing a load when settings_load is called.

rakons, 2019-06-07:
I keep saying that this is overkill to enable and disable selected handlers in structures - then why you need to pass a key to load to the settings_load if you already enabled only selected handler, and this limits the possibility to move the setting handlers to ROM in the future. But I wish to see it implemented to judge. 
And no - I was not planning to add anything similar. The fact if something has to be loader really can be judged inside the handler. But let's not talk about these ideas until some implementation is already provided. As we may think about totally different mechanisms.

Laczen, 2019-06-07:
> I keep saying that this is overkill to enable and disable selected handlers in structures - then why you need to pass a key to load to the settings_load if you already enabled only selected handler, [...]

@rakons, this is exactly what I am proposing, inside the settings_handler routine (I should have used set routine) not the settings_handler structure. The first thing that is done inside the set routine is that a flag is checked if it should be handled.

Laczen, 2019-06-08:
Just created #16711.

Vudentz, 2019-06-10:
Btw regarding the use of handlers, perhaps we could get rid of them completely if we make the load/save take the callback pointer which means we no longer have to do any lookups. Obviously this modify the whole architecture to work more like pull mode, like someone mentioned in the dev meeting, so the question is if we need to keep something like existing, maybe to lower the number of IOs, etc, but we can create other mechanism to deal with IO like caching loads and defering saves, though the later can cause data lost in the event of power lost, etc.

rakons, 2019-06-10:
@Vudentz - I would rather redesign the system to use real pull architecture than implementing such pull-push hybrid little monster :)

Vudentz, 2019-06-10:
> @Vudentz - I would rather redesign the system to use real pull architecture than implementing such pull-push hybrid little monster :)

Maybe Im being too optimistic, but it doesn't seems to be such a big deal once we have subtree loading it should be relatively simple, in my mind at least, to instead of doing a lookup for a handler just do the necessar IO, if any, to load the settings and call the callback(s) given (+ some user data if we find that useful).

Laczen, 2019-06-11:
The proposal of @Vudentz is indeed very easy to do: adapt csi->load so that it would accept a function (the set function) to be called when a match is found. To provide all functionality of settings_load_subtree() it would be required to provide:
a. The set routine,
b. The name that is considered the start of the variable names,
c. The subtree that puts an extra filer on the variable names,
Then we can remove all lookups for setting_handlers.

rakons, 2019-06-11:
@Laczen @Vudentz 
And then connect it all with settings injection.
It isn't hard to implement. It just seems messy in usage.

But if you really wish to have it - show it in implementation and unit-test/sample, then we may discuss it.

Laczen, 2019-06-13:
Since yesterday during the meeting it was clear that the next step would be to put everything in ROM, here is some code how this could be done (I have been misusing the _bt_settings_handler region in ROM):
```
/*
 * ROMSETTINGS for Zephyr
 *
 * Copyright (c) 2019 Laczen
 *
 * SPDX-License-Identifier: Apache-2.0
 */

#include <zephyr.h>
#include <string.h>
#include <settings/settings.h>

struct bt_settings_handler {
	const char *name;
	int (*get)(const char *name, size_t len, settings_read_cb read_cb,
		   void *cb_arg);
	int (*set)(const char *name, size_t len, settings_read_cb read_cb,
		   void *cb_arg);
	int (*commit)(void);
	int (*export)(int (*func)(const char *name,
				  const void *val, size_t val_len));
};

#define SETTINGS_DEFINE(_name, _get, _set, _commit, _export)         	\
	const Z_STRUCT_SECTION_ITERABLE(bt_settings_handler, 		\
	 _CONCAT(hname, __COUNTER__)) = {				\
				.name = _name,				\
				.get = _get,				\
				.set = _set,                            \
				.commit = _commit,                      \
				.export = _export,                      \
			}

SETTINGS_DEFINE("test", NULL, NULL, NULL, NULL);
SETTINGS_DEFINE("test/bb", NULL, NULL, NULL, NULL);

/* Linker-defined symbols bound to the bt_settings_handler structs */
extern const struct bt_settings_handler _bt_settings_start[];
extern const struct bt_settings_handler _bt_settings_end[];

const struct bt_settings_handler *bt_settings_parse_and_lookup(const char *name,
						   const char **next)
{
	const struct bt_settings_handler *h, *bestmatch;
	const char *tmpnext;

	bestmatch = NULL;
	if (next) {
		*next = NULL;
	}

	for (h = _bt_settings_start; h < _bt_settings_end; h++) {
		if (settings_name_steq(name, h->name, &tmpnext)) {
			if (!bestmatch) {
				bestmatch = h;
				if (next) {
					*next = tmpnext;
				}
			} else {
				if (strcmp(bestmatch->name, h->name) < 0) {
					bestmatch = h;
					if (next) {
						*next = tmpnext;
					}
				}
			}
		}
	}
	return bestmatch;
}

void main(void)
{
	const struct bt_settings_handler *h1, *h2, *h3;
	const char *next;
	const char test1[]="test/aa";
	const char test2[]="test/bb/aa";

	h1 = bt_settings_parse_and_lookup(test1, &next);
	printk("handler name: %s, next: %s\n", h1->name, next);
	h2 = bt_settings_parse_and_lookup(test2, &next);
	printk("handler name: %s, next: %s\n", h2->name, next);
	h3 = bt_settings_parse_and_lookup(test1, &next);
	printk("handler name: %s, next: %s\n", h3->name, next);

}
```

closed 2019-06-20 by Laczen