zephyrproject-rtos/zephyr #16687: basic disco sample fails

FRASTM, 2019-06-07:
basic/disco example fails when building on stm32f469i discovery board
LED1 is not defined 

```
$ west build -b stm32f469i_disco samples/basic/disco/
source directory: ./bin/zephyrproject/zephyr/samples/basic/disco
build directory: ./bin/zephyrproject/zephyr/build (created)
BOARD: stm32f469i_disco (origin: command line)
Zephyr version: 1.14.99
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.5.2", minimum required is "3.4") 
-- Selected BOARD stm32f469i_disco
-- Found west: ./.local/bin/west (found suitable version "0.5.7", minimum required is "0.5.6")
-- Loading ./bin/zephyrproject/zephyr/boards/arm/stm32f469i_disco/stm32f469i_disco.dts as base
-- Overlaying ./bin/zephyrproject/zephyr/dts/common/common.dts
Parsing Kconfig tree in ./bin/zephyrproject/zephyr/Kconfig
Loading ./bin/zephyrproject/zephyr/boards/arm/stm32f469i_disco/stm32f469i_disco_defconfig as base
Merging ./bin/zephyrproject/zephyr/samples/basic/disco/prj.conf
Configuration written to './bin/zephyrproject/zephyr/build/zephyr/.config'
-- Cache files will be written to: ./.cache/zephyr
-- The C compiler identification is GNU 8.3.0
-- The CXX compiler identification is GNU 8.3.0
-- The ASM compiler identification is GNU
-- Found assembler: ./bin/zephyr-sdk/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc
-- Performing Test toolchain_is_ok
-- Performing Test toolchain_is_ok - Success
Including module: tinycbor in path: ./bin/zephyrproject/modules/lib/tinycbor
Including module: qmsi in path: ./bin/zephyrproject/modules/hal/qmsi
Including module: esp-idf in path: ./bin/zephyrproject/modules/hal/esp-idf/zephyr
Including module: fatfs in path: ./bin/zephyrproject/modules/fs/fatfs
Including module: cypress in path: ./bin/zephyrproject/modules/hal/cypress
Including module: silabs in path: ./bin/zephyrproject/modules/hal/silabs
Including module: libmetal in path: ./bin/zephyrproject/modules/hal/libmetal
Including module: mcumgr in path: ./bin/zephyrproject/modules/lib/mcumgr
Including module: nffs in path: ./bin/zephyrproject/modules/fs/nffs
Including module: open-amp in path: ./bin/zephyrproject/modules/lib/open-amp
Including module: segger in path: ./bin/zephyrproject/modules/debug/segger
-- Configuring done
-- Generating done
-- Build files have been written to: ./bin/zephyrproject/zephyr/build
[3/115] Preparing syscall dependency handling

[13/115] Building C object CMakeFiles/app.dir/src/main.c.obj
FAILED: ccache ./bin/zephyr-sdk/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc -DBUILD_VERSION=zephyr-v1.14.0-1330-g442df9736641 -DKERNEL -DSTM32F469xx -DUSE_FULL_LL_DRIVER -DUSE_HAL_DRIVER -D_FORTIFY_SOURCE=2 -D__ZEPHYR__=1 -I./bin/zephyrproject/zephyr/kernel/include -I./bin/zephyrproject/zephyr/arch/arm/include -I./bin/zephyrproject/zephyr/include -I./bin/zephyrproject/zephyr/include/drivers -Izephyr/include/generated -I./bin/zephyrproject/zephyr/soc/arm/st_stm32/stm32f4 -I./bin/zephyrproject/zephyr/lib/libc/minimal/include -I./bin/zephyrproject/zephyr/drivers -I./bin/zephyrproject/zephyr/ext/hal/cmsis/Include -I./bin/zephyrproject/zephyr/ext/hal/st/stm32cube/stm32f4xx/soc -I./bin/zephyrproject/zephyr/ext/hal/st/stm32cube/stm32f4xx/drivers/include -I./bin/zephyrproject/zephyr/ext/hal/st/stm32cube/stm32f4xx/drivers/include/Legacy -isystem ./bin/zephyr-sdk/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/8.3.0/include -isystem ./bin/zephyr-sdk/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/8.3.0/include-fixed -Os -nostdinc -g -imacros ./bin/zephyrproject/zephyr/build/zephyr/include/generated/autoconf.h -ffreestanding -fno-common -mthumb -mcpu=cortex-m4 -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -Wno-pointer-sign -Wpointer-arith -Wno-unused-but-set-variable -Werror=implicit-int -fno-asynchronous-unwind-tables -fno-pie -fno-pic -fno-strict-overflow -fno-reorder-functions -fno-defer-pop -fmacro-prefix-map=./bin/zephyrproject/zephyr=. -ffunction-sections -fdata-sections -mabi=aapcs -march=armv7e-m -std=c99 -MD -MT CMakeFiles/app.dir/src/main.c.obj -MF CMakeFiles/app.dir/src/main.c.obj.d -o CMakeFiles/app.dir/src/main.c.obj   -c ./bin/zephyrproject/zephyr/samples/basic/disco/src/main.c
./bin/zephyrproject/zephyr/samples/basic/disco/src/main.c: In function 'main':
./bin/zephyrproject/zephyr/samples/basic/disco/src/main.c:25:29: error: 'LED1_GPIO_CONTROLLER' undeclared (first use in this function)
  gpio1 = device_get_binding(PORT1);
                             ^~~~~~~             
./bin/zephyrproject/zephyr/samples/basic/disco/src/main.c:25:29: note: each undeclared identifier is reported only once for each function it appears in
./bin/zephyrproject/zephyr/samples/basic/disco/src/main.c:28:28: error: 'LED1_GPIO_PIN' undeclared (first use in this function)
  gpio_pin_configure(gpio1, LED1, GPIO_DIR_OUT);
                            ^~~~~~~~~~~~~
[13/115] Building C object zephyr/CMakeFiles/zephyr.dir/arch/common/sw_isr_common.c.obj
ninja: build stopped: subcommand failed.
ERROR: command exited with status 1: ./bin/cmake/cmake-3.13.1-Linux-x86_64/bin/cmake --build ./bin/zephyrproject/zephyr/build

```

rgf97, 2019-06-07:
I confirm

jfischer-phytec-iot, 2019-06-07:
@FRASTM There is no led1 alias, you could fix it easy

```diff
diff --git a/boards/arm/stm32f469i_disco/stm32f469i_disco.dts b/boards/arm/stm32f469i_disco/stm32f469i_disco.dts
index a7f9e033bf..523a6e7fa9 100644
--- a/boards/arm/stm32f469i_disco/stm32f469i_disco.dts
+++ b/boards/arm/stm32f469i_disco/stm32f469i_disco.dts
@@ -49,6 +49,7 @@
 
        aliases {
                led0 = &green_led_1;
+               led1 = &orange_led_2;
                sw0 = &user_button;
        };
 };

```

closed 2019-07-03 by erwango