zephyrproject-rtos/zephyr #16707: Problem with k_sleep

pmishr16, 2019-06-08:
I  created a shell command which invokes two threads: producer and consumer.However after going to sleep due to k_sleep(10) function, the threads don't wake up.The print statement after the k_sleep function does not execute and the while loop runs only once.Kindly suggest the solution

static int cmd_task2(const struct shell *shell, size_t argc, char **argv)
{
	ARG_UNUSED(argc);
	ARG_UNUSED(argv);
        k_sem_init(&my_sem, 1, 1);
        
        
        k_msgq_init(&my_msgq, my_msgq_buffer, sizeof(temp), 10);
        producer_tid = k_thread_create(&my_thread_data, a_stack_area,
                                 MY_STACK_SIZE,
                                 producer,
                                 NULL, NULL, NULL,
                                 PRIORITY, 0, K_NO_WAIT);
        k_thread_name_set(producer_tid, "thread_producer");
        consumer_tid = k_thread_create(&my_thread_data, b_stack_area,
                                 MY_STACK_SIZE,
                                 consumer,
                                 NULL, NULL, NULL,
                                 PRIORITY, 0, K_NO_WAIT);
        k_thread_name_set(consumer_tid, "consumer_thread");
      return 0;
}

extern void producer(void * dummy1, void * dummy2, void * dummy3);
extern void consumer(void * temp1, void * temp2, void * temp3);
void producer(void * dummy1,void * dummy2, void * dummy3){
   ARG_UNUSED(dummy1);
	ARG_UNUSED(dummy2);
	ARG_UNUSED(dummy3);
    
    struct data_item_type data;
    
    int count=0;  
      while(count<1000){ 
          printk("start\n");
          if(k_sem_take(&my_sem, K_FOREVER)!=0){
              printk("semaphore for producer not available\n");
          }
          data.field1=k_cycle_get_32();
          while (k_msgq_put(&my_msgq, &data, K_NO_WAIT) != 0) {
            /* message queue is full: purge old data & try again */
            k_msgq_purge(&my_msgq);
        }
          printk("The data send is  %"PRIu32"  \n", data.field1);
         k_sem_give(&my_sem);
         k_sleep(10);
         printk("wake");
      }
   return;
}
void consumer(void * temp1,void * temp2, void * temp3){
ARG_UNUSED(dummy1);
	ARG_UNUSED(dummy2);
	ARG_UNUSED(dummy3);
    struct data_item_type data;
    int count=0;
    while(count<1000){ 
            printk("end\n");
            if(k_sem_take(&my_sem, K_FOREVER)!=0){
              printk("semaphore for consumer not available\n");
           }else{
        k_msgq_get(&my_msgq, &data, K_FOREVER);
        printk("the data received is %"PRIu32"  \n",data.field1);
        k_sem_give(&my_sem);
        k_sleep(10);
        printk("wake");
       }
    }
    return;
}

pmishr16, 2019-06-08:
#include <zephyr.h>
//#include <board.h>
#include <version.h>
#include <device.h>
#include <gpio.h>
#include <pwm.h>
#include <misc/util.h>
#include <misc/printk.h>
#include <pinmux.h>
#include "../boards/x86/galileo/board.h"
#include "../boards/x86/galileo/pinmux_galileo.h"
#include "../drivers/gpio/gpio_dw_registers.h"
#include "../drivers/gpio/gpio_dw.h"
#include <shell/shell.h>
#include <inttypes.h>

#define EDGE_FALLING    (GPIO_INT_EDGE | GPIO_INT_ACTIVE_LOW)
#define EDGE_RISING	(GPIO_INT_EDGE | GPIO_INT_ACTIVE_HIGH)
#define PULL_UP 	0	/* change this to enable pull-up/pull-down */
#define SLEEP_TIME	1000	/* Sleep time */

int intno;		// numver of interrupts
int time1,time2;
u32_t task1[100];
int count=0;
struct device *gpiob_1;
struct k_sem my_sem;

#define MY_STACK_SIZE 500
#define PRIORITY 5

struct data_item_type {
    u32_t field1;
} temp;
K_THREAD_STACK_DEFINE(b_stack_area, MY_STACK_SIZE);
K_THREAD_STACK_DEFINE(a_stack_area, MY_STACK_SIZE);
char __aligned(4) my_msgq_buffer[10 * sizeof(temp)];
struct k_thread my_thread_data;
struct k_msgq my_msgq;
k_tid_t producer_tid,consumer_tid;
extern void producer(void * dummy1, void * dummy2, void * dummy3);
extern void consumer(void * temp1, void * temp2, void * temp3);
void producer(void * dummy1,void * dummy2, void * dummy3){
   ARG_UNUSED(dummy1);
	ARG_UNUSED(dummy2);
	ARG_UNUSED(dummy3);
    
    struct data_item_type data;
    
    int count=0;  
      while(count<1000){ 
          printk("start\n");
          if(k_sem_take(&my_sem, K_FOREVER)!=0){
              printk("semaphore for producer not available\n");
          }
          data.field1=k_cycle_get_32();
          while (k_msgq_put(&my_msgq, &data, K_NO_WAIT) != 0) {
            /* message queue is full: purge old data & try again */
            k_msgq_purge(&my_msgq);
        }
          printk("The data send is  %"PRIu32"  \n", data.field1);
         k_sem_give(&my_sem);
         k_sleep(10);
         printk("wake");
      }
   return;
}
void consumer(void * dummy1,void * dummy2, void * dummy3){
ARG_UNUSED(dummy1);
	ARG_UNUSED(dummy2);
[... 221 lines cut ...]
		gpio_pin_write(gpiob, 5, (iter & BIT(2))>> 2);
		
		gpio_pin_read(gpiob, 6, &val);

		ret = gpio_port_read(gpiob, &gpio_d);
		
		printk("iter and gpio_0  %x  %x, %x, %x, %x\n", iter, val, base_addr, gpio_d, intno);

		total_iter++; 
		iter++;
		if (iter ==8) 	{
			iter=0;
			if (!flag) { 
				ret = pwm_pin_set_cycles(pwm0, 3, 0, 3000); // switch to a large duty cycle
				ret = pwm_pin_set_cycles(pwm0, 7, 0, 3000);
				flag=1;}
			else { 
				ret = pwm_pin_set_cycles(pwm0, 3, 0, 200); 	// switch to a small duty cycle
				ret = pwm_pin_set_cycles(pwm0, 7, 100, 0);		// turn off pwm 7
				flag=0;}	
		}
		k_sleep(SLEEP_TIME);
	}*/
 
}`

andyross, 2019-06-08:
What platform are you on?  Can you attach (not paste) the relevant code, including build integration and kconfig configuration so I can try to reproduce?  Can you make the failure happen independently of the shell stuff to simplify the test case?

andyross, 2019-08-14:
Any update?  I'm still confused about what the problem is here and what you expect should be happening.  And the code as it is appears to be specific to galileo, so I can't run it to experiment myself.

ioannisg, 2019-08-19:
@pmishr16 could you respond here, please?