zephyrproject-rtos/zephyr #16716: soc: stm32: Is the setting of NUM_IRQS in the F3 series wrong?

KwonTae-young, 2019-06-10:
I am currently looking at the soc settings of the `F3` series.
The current `F3` series are all set to `82` with the same `NUM_IRQS`.
``` 
config NUM_IRQS
	int
	default 82
```

As far as I know, these settings seem to be wrong.

**STM32F302X8**
Current code: https://github.com/zephyrproject-rtos/zephyr/blob/master/soc/arm/st_stm32/stm32f3/Kconfig.defconfig.stm32f302x8#L18-L20
Datasheet: https://www.st.com/resource/en/datasheet/stm32f302c8.pdf
[image]

**STM32F303XC**
Current code: https://github.com/zephyrproject-rtos/zephyr/blob/master/soc/arm/st_stm32/stm32f3/Kconfig.defconfig.stm32f303xc#L18-L20
Datasheet: https://www.st.com/resource/en/datasheet/stm32f303vc.pdf
[image]

**STM32F334X8**
Current code: https://github.com/zephyrproject-rtos/zephyr/blob/master/soc/arm/st_stm32/stm32f3/Kconfig.defconfig.stm32f334x8#L18-L20
Datasheet: https://www.st.com/resource/en/datasheet/stm32f334r8.pdf
[image]

**STM32F373XC**
Current code: https://github.com/zephyrproject-rtos/zephyr/blob/master/soc/arm/st_stm32/stm32f3/Kconfig.defconfig.stm32f373xc#L18-L20
Datasheet: https://www.st.com/resource/en/datasheet/stm32f373c8.pdf
[image]

Is the setting of `NUM_IRQS` of F3 series wrong?
Or am I mistaken about the setting value of `NUM_IRQS`?

I'm sorry I do not have enough English. :)
Thank you.

ioannisg, 2019-06-10:
@erwango could you address this one?

KwonTae-young, 2019-06-11:
I reviewed all the STM32 series and created a commit to modify `NUM_IRQS`.

ydamigos, 2019-06-11:
@KwonTae-young Please check IRQn_Type enumerations in SoC header files at `ext/hal/st/stm32cube/stm32fxxx/soc`. `NUM_IRQS` is equal to the maximum IRQ number plus 1.

KwonTae-young, 2019-06-11:
@ydamigos Thank you. I was misunderstanding NUM_IRQS. :sweat_smile:

closed 2019-06-11 by KwonTae-young