zephyrproject-rtos/zephyr #16721: PCIe build warnings from devicetree

gnuless, 2019-06-10:
**Describe the bug**

When building targets with PCIe devices in devicetree, warnings are produced that need to be silenced.

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=up_squared
3. make
4. See warnings like `up_squared.dts.pre.tmp:94.15-105.5: Warning (simple_bus_reg): /soc/i2c@1: simple-bus unit address format error, expected "b100"`

**Impact**
It's an annoyance (and bad form).

gnuless, 2019-06-21:
@galak and @ulfalizer are rewriting the devicetree scripts. Waiting on that before I fix this.

gnuless, 2019-08-25:
Changed to medium-priority bug to be sure this stays on our radar for 2.0 release.

ioannisg, 2019-08-26:
@gnuless pls, give it a try if you have time :p

gnuless, 2019-09-06:
@galak can you revisit this now that the devicetree stuff has been rewritten? the new Python scripts double the number of warnings spewed during the configuration process. I'm really not sure I want to start tinkering with work-in-progress devicetree code, but I imagine restructuring the .dts files and emitting b/d/f for PCI devices can be all that hard.

closed 2019-10-18 by gnuless