zephyrproject-rtos/zephyr #16739: spi: stm32: pinmux: default configuration does not opt for low power consumption on deep sleep

gon1332, 2019-06-11:
**Is your enhancement proposal related to a problem? Please describe.**
The SPI MOSI/MISO are configured as PP NOPULL from the pinmux. The use case is when having a SPI peripheral. Putting the board in STOP2 mode with the current configuration, the consumption is approximately 20uA (per pin) more than having those pins set as PP PULLUPs.

**Describe the solution you'd like**
I would like to have these pins configured as PP PULLUPS from the pinmux driver.

**Describe alternatives you've considered**
Set these pins as PULLUPS after Zephyr initialization from the application.

**Additional context**
Tested on a custom board with a STM32L496RE and on Nucleo L496ZG.

If you think that's OK, I can open a PR to fix this.

gon1332, 2019-06-11:
@galazari @nikooiko

gon1332, 2019-06-11:
Also the same behaviour is observed while in RUN mode.

erwango, 2019-07-26:
@gon1332 , I'm fine with this change

loicpoulain, 2019-07-30:
A bus like SPI is generally push-pull, I don't fully understand the root cause of this issue, any input or official recommendation from ST is welcome here. Moreover, I suspect adding a pull (up/down) increases consumption during active transfers and **may** cause issue at highest speeds due to the created low pass filter, though it needs to be calculated.

erwango, 2019-07-30:
One interesting document could be found here:
https://www.st.com/resource/en/application_note/dm00033348.pdf
Though it does not mention anything on pupd pin configuration.
One interesting hint though: deactivate GPIO clock once pin configuration is done.

loicpoulain, 2019-07-31:
> One interesting document could be found here:
> https://www.st.com/resource/en/application_note/dm00033348.pdf
> Though it does not mention anything on pupd pin configuration.
> One interesting hint though: deactivate GPIO clock once pin configuration is done.

Indeed, it deserves a try!

PR #17839 (2019-08-01): stm32: pinmux: spi: Opt for lower power consumption
Fixes #16739. Changed the STM32's SPI MISO/MOSI configurations in order
to reduce the power consumption by approximately 20uA per pin.
commit dca08c95 (2019-08-01): stm32: pinmux: spi: Opt for lower power consumption

Fixes #16739. Changed the STM32's SPI MISO/MOSI configurations in order
to reduce the power consumption by approximately 20uA per pin.

According to STM32's Application Notes on GPIO configuration for
low-power consumption, the input pins should be configured with internal
pull-up or pull-down resistor. If a pin is configured as a floating
input, and there is no signal present, the Schmitt trigger randomly
toggles between the logical levels induced by the external noise, thus
increasing the consumption.
closed 2019-08-01 by galak