zephyrproject-rtos/zephyr #16760: K_THREAD_STACK_EXTERN() confuses gen_kobject_list.py

andrewboie, 2019-06-11:
**Describe the bug**
If a C file is built that has an extern declaration of a thread stack, before the actual declaration of the thread stack, the kernel object detection mechanism will fail to register the stack object and system calls on it (such as k_thread_create()) will fail:

```
0x00423000 is not a valid _k_thread_stack_element
syscall z_hdlr_k_thread_create failed check: bad stack object
***** Kernel OOPS! *****
Current thread ID = 0x00401ae0
eax: 0x00000000, ebx: 0x00000000, ecx: 0x00000000, edx: 0x00000000
esi: 0x00000000, edi: 0x00000000, ebp: 0x00000000, esp: 0x00428f98
eflags: 0x00000246 cs: 0x002b
call trace:
eip: 0x00002fb0
NULL base ptr
Fatal fault in thread 0x00401ae0! Aborting.
```

This could happen commonly if a header file has the extern declaration, and the C file which defines the thread stack object includes it.

**To Reproduce**
Build an application with CONFIG_USERSPACE enabled with the following code snippet:

```
K_THREAD_STACK_EXTERN(ustack);
K_THREAD_STACK_DEFINE(ustack, STACK_SIZE);
```

If you build with VERBOSE=1, the output of gen_kobject_list.py will not show it.

If you then comment out the K_THREAD_STACK_EXTERN() line, it will be detected:

`
gen_kobject_list.py: symbol 'ustack' at 0x423000 contains 1 object(s)
`

**Expected behavior**
gen_kobject_list.py should not disregard thread stacks which have an extern declaration in the same compilation unit.

**Impact**
major annoyance, possible to work around. will greatly confuse users who don't know about this bug.

please backport to 1.14

**Environment (please complete the following information):**
Seen on qemu_x86, probably affects all userspace arches

andrewboie, 2019-06-11:
@wentongwu can you take a look at this one? This would be a good opportunity to learn more about how gen_kobject_list.py works.

wentongwu, 2019-06-12:
> @wentongwu can you take a look at this one? This would be a good opportunity to learn more about how gen_kobject_list.py wo

wentongwu, 2019-06-12:
sorry, mis-click.
yes, I'm checking this issue. Thanks

wentongwu, 2019-06-16:
will add analysis for DW_TAG_typedef to fix this issue.

wentongwu, 2019-08-16:
not only DW_TAG_typedef doesn't been considered, but also handle "extern type variable[]" incorrectly.

PR #18710 (2019-08-29): scripts: elf_helper.py: add analyze for DW_TAG_typedef
1) add analyze for DW_TAG_typedef in order to catch the kernel objects with typedef 
2) If a stack is declared with K_THREAD_STACK_EXTERN first, analyze array in elf_helper.py will ignore this declaration which will be referenced by the actual instances via the tag DW_AT_specification, so that this stack can't be detected by the kernel object detection mechanism, and this will cause user-space not work.
 

Fixes: #16760.
commit 9ca0d5e1 (2019-08-29): scripts: elf_helper.py: add analyze for DW_TAG_typedef

add analyze for DW_TAG_typedef in order to catch all the
kernel objects.

Fixes: #16760.
commit ed6ce291 (2019-08-29): scripts: elf_helper.py: fix stack declared K_THREAD_STACK_EXTERN

If a stack is declared with K_THREAD_STACK_EXTERN first, analyze array
in elf_helper.py will ignore this declaration which will be referenced
by the actual instances via the tag DW_AT_specification, so that this
stack can't be detected by the kernel object detection mechanism, and
this will cause userspace not work.

Fixes: #16760.
commit f86fe3ff (2019-08-29): scripts: elf_helper.py: add analyze for DW_TAG_typedef

add analyze for DW_TAG_typedef in order to catch all the
kernel objects.

Fixes: #16760.
closed 2019-08-29 by nashif