zephyrproject-rtos/zephyr #16767: net: net_pkt_clone

alexanderwachter, 2019-06-12:
**net_pkt_clone doesn't change the lladdr pointer to the new packet**
`net_pkt_clone` copies the `lladdr_src` and `lladdr_dst` from the source to the clone.
Usually those link layer address points to the buffer from the net_pkt itself. When the original pkt is de-referenced  those addresses become invalid.
**The clone is not an exact copy of the source**
The clone is not an exact copy because data that was pulled before is not taken into account. Ethernet for example pulls the addresses and therefor they are not copied by `net_pkt_copy`.

**Describe the solution you'd like**
 - Maybe a check for  'destination is inside the buffer' and if it's true, move the destination to the new buffer. This will only works when the address in the source pkt was not pulled before!
 - Maybe net_pkt_copy should have a flag for an exact copy.

kartben, 2023-09-19:
I think this was fixed with #51266 and commit 389bd577536a3a3e783ea3fa1d2cf9b9f31541ac -- @fgrandel to confirm

fgrandel, 2023-09-19:
@kartben Absolutely - didn't know about this issue. IMO can be closed as fixed.

closed 2023-09-19 by kartben