zephyrproject-rtos/zephyr #16773: DTS: generated output for each flash-controller

nvlsianpu, 2019-06-12:
**Problem to be solved**
Currently flash-controller is generated only for `zephyr,flash` chosen node.
This causes problem when need to use DTS data form more than one flash controller at one.

For instance when we want to use flash-simulator driver (on qemu_x86), then need to chose its DT node. This imposes that its values are presented as `DT_FLASH_*` defines. As the result target has fake ROM size etc.

**solution proposal**
Generate substitute of flowing DT macros;
```
DT_FLASH_BASE_ADDRESS
DT_FLASH_ERASE_BLOCK_SIZE
DT_FLASH_WRITE_BLOCK_SIZE
DT_FLASH_SIZE
DT_FLASH_DEV_NAME
```
for each flash-controller node. Macros for chosen flash flash-controller might be duplicated.

Looks like generation of above values based on scripts hacks, not on yaml bindings - which makes it more difficult.

**Context**
Situation we have are source of issue signaled https://github.com/zephyrproject-rtos/zephyr/issues/15831 https://github.com/zephyrproject-rtos/zephyr/issues/15831#issuecomment-498771321

nvlsianpu, 2019-06-12:
cc @anangl , @carlescufi , @galak , @ulfalizer , @b0661  Need someone to deal with this issue.

galak, 2019-06-12:
Are you looking for `SOC_NV_FLASH` defines to get generated for both flash nodes in qemu_x86.dts?

nvlsianpu, 2019-06-14:
I see that there are SOC_NV_FLASH_xxx defines. But they are useless for this case as can't predict whether flash-simulator will have index 0 or 1 (SOC_NV_FLASH_0_xxx or SOC_NV_FLASH_1_xxx belongs to flash-sim).

What I want to have is unique defines for runing flash-simulator without chosen flash_sim0 node:
```
chosen {
		zephyr,flash = &flash_sim0;
	};
```

galak, 2019-07-03:
It sounds as if the flash simulator should have a different compatible then.

nvlsianpu, 2019-07-03:
^^ Can you propose solution?

galak, 2019-07-03:
> ^^ Can you propose solution?

I'm still not clear on what the issue is, or what defines you need.

nvlsianpu, 2019-07-03:
DT_FLASH_SIM_BASE_ADDRESS
DT_FLASH_SIM_ERASE_BLOCK_SIZE
DT_FLASH_SIM_WRITE_BLOCK_SIZE
DT_FLASH_SIM_SIZE
DT_FLASH_SIM_DEV_NAME

galak, 2019-07-03:
You have:

```
#define DT_SOC_NV_FLASH_0_BASE_ADDRESS                  0x0
#define DT_SOC_NV_FLASH_0_ERASE_BLOCK_SIZE              1024
#define DT_SOC_NV_FLASH_0_SIZE                          1048576
#define DT_SOC_NV_FLASH_0_WRITE_BLOCK_SIZE              4
```

nvlsianpu, 2019-07-03:
For modified qemu_x86.dts

```
	chosen {
		zephyr,sram = &sram0;
		/*zephyr,flash = &flash_sim0;*/
		zephyr,flash = &flash0;
		zephyr,console = &uart0;
		zephyr,shell-uart = &uart0;
		zephyr,bt-uart = &uart1;
		zephyr,uart-pipe = &uart1;
		zephyr,bt-mon-uart = &uart1;
		zephyr,code-partition = &slot0_partition;
	};

	soc {
		eth0: eth@febc0000 {
			compatible = "intel,e1000";
			reg = <0xfebc0000 0x100>;
			label = "eth0";
			interrupts = <11 IRQ_TYPE_EDGE_RISING 3>;
			interrupt-parent = <&intc>;

			status = "okay";
		};
	};

	sim_flash {
		compatible = "zephyr,sim-flash";
		label = "FLASH_SIMULATOR";

		#address-cells = <1>;
		#size-cells = <1>;

		flash_sim0: flash_sim@0 {
			compatible = "soc-nv-flash";
			reg = <0x00000000 DT_SIZE_K(1024)>;

			erase-block-size = <1024>;
			write-block-size = <4>;
		};
	};
```
I've got in DTS product:
```
/* sim_flash */
#define DT_ZEPHYR_SIM_FLASH_0			1
#define DT_ZEPHYR_SIM_FLASH_SIM_FLASH_LABEL	"FLASH_SIMULATOR"
#define DT_INST_0_ZEPHYR_SIM_FLASH_LABEL	DT_ZEPHYR_SIM_FLASH_SIM_FLASH_LABEL
#define DT_ZEPHYR_SIM_FLASH_0_LABEL __DEPRECATED_MACRO DT_ZEPHYR_SIM_FLASH_SIM_FLASH_LABEL

/* flash_sim@0 */
#define DT_SOC_NV_FLASH_0				1
#define DT_SOC_NV_FLASH_0_BASE_ADDRESS			0x0
#define DT_SOC_NV_FLASH_0_ERASE_BLOCK_SIZE		1024
#define DT_SOC_NV_FLASH_0_SIZE				1048576
#define DT_SOC_NV_FLASH_0_WRITE_BLOCK_SIZE		4
#define DT_INST_0_SOC_NV_FLASH_BASE_ADDRESS		DT_SOC_NV_FLASH_0_BASE_ADDRESS
#define DT_INST_0_SOC_NV_FLASH_ERASE_BLOCK_SIZE		DT_SOC_NV_FLASH_0_ERASE_BLOCK_SIZE
#define DT_INST_0_SOC_NV_FLASH_SIZE			DT_SOC_NV_FLASH_0_SIZE
#define DT_INST_0_SOC_NV_FLASH_WRITE_BLOCK_SIZE		DT_SOC_NV_FLASH_0_WRITE_BLOCK_SIZE
```

@galak I have quite a wide range of areas on my head, so excuse basic question:
Should i just use `DT_SOC_NV_FLASH_0_xxx` macros. for obtain values i want? Is index `0` guaranted?
Looks like you points this before - but I didn't catch.

galak, 2019-07-03:
The `DT_SOC_NV_FLASH_0_XXX` are not index based, but based on node name/address the flash is at.

nvlsianpu, 2019-07-03:
Understand: this mean that this might works - will experiment with this.

Strange thing- now even if i removed sim_flash node completly form dts, and remove flash simulator from the compilation, build failure in a strange way:

```
[62/71] Linking C executable zephyr/zephyr_prebuilt.elf
Memory region         Used Size  Region Size  %age Used
             ROM:       19736 B         4 MB      0.47%
             RAM:          8 KB         4 MB      0.20%
        IDT_LIST:         416 B         2 KB     20.31%
        MMU_LIST:         104 B         1 KB     10.16%
[68/71] Generating mmu_tables.bin
FAILED: zephyr/arch/arch/x86/mmu_tables.bin 
cd /home/anpu/codedev/zephyrproject/build/hellow/zephyr/arch/arch/x86 && /usr/bin/python3 /home/anpu/codedev/zephyrproject/zephyr/arch/x86/gen_mmu_x86.py -k /home/anpu/codedev/zephyrproject/build/hellow/zephyr/zephyr_prebuilt.elf -o mmu_tables.bin -u user_mmu_tables.bin
Memory region 1 (4000:401000) overlaps memory region 0 (400000:800000)
ninja: build stopped: subcommand failed.
ERROR: command exited with status 1: /home/anpu/cmake/cmake-3.13.1-Linux-x86_64/bin/cmake --build /home/anpu/codedev/zephyrproject/build/hellow
run as "west -v build -d build/hellow --board qemu_x86 zephyr/samples/hello_world/" for a stack trace
```

andrewboie, 2019-07-03:
Adding bug tag as this is blocking https://github.com/zephyrproject-rtos/zephyr/issues/15831

andrewboie, 2019-07-04:
> Memory region 1 (4000:401000) overlaps memory region 0 (400000:800000)

Grep for MMU_BOOT_REGION in the code, instances of these are used to construct the page table. Somewhere, these are overlapping. `make VERBOSE=1` may have more clues.

nvlsianpu, 2019-07-04:
^^ ROM (Flash) region overlaps RAM region.

nvlsianpu, 2019-07-04:
hopefully fix in a way by #17349

ioannisg, 2019-07-18:
@nvlsianpu #17349 is now merged, so can you revisit this one?

galak, 2019-07-25:
> @nvlsianpu #17349 is now merged, so can you revisit this one?

Is this still an open issue?

nvlsianpu, 2019-07-29:
^^ yes, this was covered in PR mentioned above.

closed 2019-07-29 by nvlsianpu