zephyrproject-rtos/zephyr #16778: Build failures in various mimxrt boards

jukkar, 2019-06-12:
**Describe the bug**
I am seeing this error in several PRs that I have submitted
```
ERROR: section nocache is 128 bytes in the virtual/runtime address space, but only 0 in the loaded/XIP section!
```
when sanitychecker uses board mimxrt1050_evk_qspi, mimxrt1050_evk or mimxrt1064_evk.
Some of the PRs that show this issue are #16743 #16464  #16403
One example shippable log can be found here
https://app.shippable.com/github/zephyrproject-rtos/zephyr/runs/43576/summary/console

**Impact**
This is a showstopper for those PRs that I mentioned above. I do not see any issue in the PRs that would cause this issue.

ioannisg, 2019-06-12:
I think I saw this recently. I'll take a look.

agansari, 2019-06-13:
@ioannisg i remember somethin similar in #15836

jukkar, 2019-06-13:
I submitted #16811 that will disable Ethernet so that networking PRs can be merged. That is of course a workaround and should be reverted eventually.

ioannisg, 2019-06-13:
@agansari yes, you're right, that's  why this looked familiar. @jukkar perhaps, the root cause **is** the check_link_map.py script. Could we confirm that before merging the revert PR?

jukkar, 2019-06-13:
> Could we confirm that before merging the revert PR?

Sure, np.

jukkar, 2019-06-18:
@ioannisg have you managed to figure out the culprit of this one?

ioannisg, 2019-06-19:
@jukkar on my task list for today :)

PR #16923 (2019-06-19): arch: common: nocache: fix linker section definition
No-cache SRAM section is currently used for ARM-only builds
with support for no-cacheable memory sections (i.e.
CONFIG_ARCH_HAS_NOCACHE_MEMORY_SUPPORT) and it holds
uninitialized data. This commit properly defines the
corresponding linker section using SECTION_DATA_PROLOGUE
and GROUP_DATA_LINK_IN macros.

The patch fixes mimxrt board compilation.

Fixes #16778
closed 2019-06-19 by andrewboie